hieplpvip/AsusSMC #2: Just a few questions...

zacharyrs, 2018-10-22:
I currently use EMlyDinEsHMG's kexts for keyboard and touchpad, but considering they haven't been updated in a while, I'm eager for an alternative. I assume that this kext would hence replace the keyboard portions, but would not help regarding the touchpad.

I'm just wondering if you hence use EMlyDinEsHMG's touchpad parts, or perhaps you could point me in the direction of an alternative?

Also, am I right to assume with this I would not need your AsusFnKeys driver?

Lastly, are the DSDT patches simply like those required by EMlyDinEsHMG's kexts, or are they different?

Thanks!

hieplpvip, 2018-10-22:
1. It seems like you're misunderstanding AsusSMC. To be clear, it has nothing to do with keyboard support. AsusSMC only adds support for ALS and keyboard backlight, as well as Fn keys. You still need a keyboard kext (eg. VoodooPS2Controller or EMlyDinEsHMG's kexts).

2. I assume you have PS/2 trackpad. EMlyDinEsHMG's kexts are the best one now, you should continue using them.

3. AsusSMC is a replacement for AsusFnKeys

4. DSDT patches are different. I don't have time to write an instruction now, sorry.

zacharyrs, 2018-10-23:
Hi, thanks for the response!

1. I apologise for the misunderstanding, but thanks for clarifying it.

2. Yes I do have a PS/2 trackpad. Guess I'll just wait it out for EMlyDinEsHMG to release the source...

3. I see, that makes sense.

4. No worries! Any chance you'd feel comfortable releasing your original and patched DSDT, so I can take a look and try reverse the patches for my own? No pressure though - if you don't want to I'll just wait it out.

hieplpvip, 2018-10-23:
Take a look at my repo for Asus Zenbook https://github.com/hieplpvip/ASUS-ZENBOOK-HACKINTOSH

zacharyrs, 2018-10-23:
Thanks for the link! Probably should've looked at your profile myself oops...

Just having had a skim, I assume the following are files of interest:
- SSDT-ALS0
- SSDT-ALSC
- SSDT-RALS
- SSDT-ATK-BDW
- SSDT-ATK-KABY

Regarding ALS0, ALSC and RALS, I have no ambient light sensor, so would I be correct to assume I don't need any of these? Or do I hence need ALS0 to fake one as per the S510?

Regarding ATK-BDW and ATK-KABY, I assume I would need to develop my own, for Haswell?
My own DSDT methods appear similar to BDW in that there's no PCI0.LPCB.EC0.ST9E method, however dissimilar in the SLKB method... Do I simply copy my own methods here?

Lastly, sorry for the repeated queries, I think I should be able to figure it out from here!

hieplpvip, 2018-10-23:
What is your laptop?

zacharyrs, 2018-10-23:
Asus GL551JW

hieplpvip, 2018-10-23:
Send me your CLVOER folder with original ACPI files

zacharyrs, 2018-10-31:
Sorry for the delay, I've been a little caught up with exams.
Attached is an archive of my clover config, and ACPI files.
The original source for my patches is found [here](https://github.com/zacharyrs/GL551JW-Hackintosh).

[CLOVER.zip](https://github.com/hieplpvip/AsusSMC/files/2535067/CLOVER.zip)

zacharyrs, 2018-11-05:
Thanks very much!

I'll give it a go tonight.

zacharyrs, 2018-11-08:
Sorry for the delay, got a little distracted...

I've just tried applying, as per the hotpatches available in my repo.
[You can view them here.](https://github.com/zacharyrs/GL551JW-Hackintosh)

Unfortunately I'm running into a little trouble with the FN keys.
Specifically when I don't apply any of the function key patches, F1, F2, F5, F6, F8, F9 don't work.
When I do apply the patches, the above still don't work, but F3 and F4 stop too.

I'm probably making a silly mistake, but any chance you could take a look?

On a sidenote, if I wanted only 4 brightness levels for the backlight, would it only be a matter of changing PWKB and the first line in the following method (Arg0 / 16 to Arg0 / 4)?

zacharyrs, 2018-11-17:
I've been trying to figure out what I might be doing wrong, and I noticed you have an I2C as opposed to PS2 trackpad.

Is this perhaps incompatible with ApplePS2SmartTouchpad? I have removed AsusNBFNKey, and disabled the related sections...

zacharyrs, 2019-01-15:
@hieplpvip 	I'll take at them soon, and see if I can spot what went wrong.
Just a little bogged down with other stuff right now.

For now I'll mark this as closed.

closed 2019-01-15 by zacharyrs