hieplpvip/AsusSMC #4: ASUS ZENBOOK UX330UAK with 4 keyboard backlight levels

anonymous-writer, 2018-11-05:
Hi hieplpvip,

Thanks for your very great work. Everything works on the above-mentioned laptop.
Only keyboard backlight do not want to work.

Should your kext also work with this kind of keyboard backlight? 
I would be very happy about your answer.

https://bitbucket.org/anonymous_writer/asus-zenbook-ux330uak-high-sierra-efi/src/master/

hieplpvip, 2018-11-05:
Of course it works.

There an installation instruction in the wiki. You should also take a look at my [Zenbook repo](https://github.com/hieplpvip/ASUS-ZENBOOK-HACKINTOSH)

anonymous-writer, 2018-11-06:
You are the best. Thanks for the new ASUS Kext.
With take over of this entry in the DSDT.aml keyboard backlight works with 16 levels too.
Everything what you write works now with the kext AsusSMC.
Thanks again.

https://github.com/hieplpvip/ASUS-ZENBOOK-HACKINTOSH/blob/master/src/hotpatch/include/SSDT-ATK-KABY.dsl

closed 2018-11-06 by anonymous-writer

anonymous-writer, 2018-11-09:
Hello @HIEPLPVIP,
Have your kext posted here. Hope this is OK for you.
Thanks again to your work.

https://www.hackintosh-forum.de/forum/thread/40125-hieplpvip-asussmc-f%C3%BCr-asus-laptops/?postID=461506#post461506