hieplpvip/AsusSMC #6: Bug chưa điều khiển được nhạc

tctien342, 2018-11-28:
Bác Hiệp có thêm function này chưa ạ, e thử các tổ Hợp FN + Mũi tên thì chưa pause track vs next track và prev track đc

hieplpvip, 2019-01-04:
@tctien342 Sorry bạn, dạo này mình bận quá. Tết có thời gian rảnh mình sẽ fix

hieplpvip, 2019-01-13:
Bạn dùng thử patch này đi:
```
# Replacing method _Q6C for media left arrow key to work
into Method label _Q6C replace_content begin 
    If (ATKP)\n
    {\n
        \_SB.ATKD.IANE (0x40)\n
    }
end;

# Replacing method _Q6D for media right arrow key to work
into Method label _Q6D replace_content begin 
    If (ATKP)\n
    {\n
        \_SB.ATKD.IANE (0x41)\n
    }
end;

# Replacing method _Q6E for media up arrow key to work
into Method label _Q6E replace_content begin 
    If (ATKP)\n
    {\n
        \_SB.ATKD.IANE (0x43)\n
    }
end;

# Replacing method _Q6F for media down arrow key to work
into Method label _Q6F replace_content begin 
    If (ATKP)\n
    {\n
        \_SB.ATKD.IANE (0x45)\n
    }
end;
```

closed 2019-07-23 by hieplpvip