hieplpvip/AsusSMC #8: DSDT patches is not even close to my DSDT

valinor61, 2019-01-14:
I have Asus GL553VD laptop it's a kabylake 7700hq device all patches in new repo seems not working on my device I installed kext also it dependencies(VirtualSMC and Lilu) I can see it on ioreg. 
Here is my DSDT can you look at it?
[DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/2757402/DSDT.aml.zip)

hieplpvip, 2019-01-15:
@valinor61 Are you sure that your laptop has keyboard backlight? I don't see any code related to keyboard backlight in your DSDT.

hieplpvip, 2019-01-15:
@valinor61 Your case is very weird, I'll do more research into it later. Now I'm pretty busy studying, sorry

valinor61, 2019-01-15:
Ok thanks I hope you can figure it out on your free time.

andreszerocross, 2019-05-03:
This notebook model use HSxx of XHC controller as keyboard interface,, not PS2.

I have installed similar ROG and generations,, when i check IOREG, keyboard is ITE xxx in HSxx port of XHC. Maybe it's the issue

hieplpvip, 2019-05-07:
@valinor61 Your laptop seems to use a different interface for keyboard backlight (which I believe something called Aura). I've found Aura port for Linux, now we just need to port it to macOS.
Will do when I have time

valinor61, 2019-05-08:
Yes Q in DSDT is not working on my device also I have no asus aura its for rgb keyboards mine use hid interface i realized that after while I hope it can be fix I have no clue for fixing it

hieplpvip, 2019-05-08:
@valinor61 Please send me output from gen_debug

hieplpvip, 2019-06-27:
#valinor61 Could you send me your ioreg report?

hieplpvip, 2019-07-23:
v2.0 added support for this laptop

closed 2019-07-23 by hieplpvip