hieplpvip/AsusSMC #9: No volume function keys on Mojave

makito89, 2019-01-18:
Hello.
I'm using an UX330UA with Skylake i7 6500U and I've successfully installed your AsusSMC.kext and have all the other function keys working (F1-Sleep, F2-Airplane mode, F3/F4-Keyboard Bakclight, F5/F6/F7-LCD Backlight, F8-LCD/External Display, F9-I2C Trackpad ON/OFF) but the F10/F11/F12 - Volume buttons do not work.
I'm using DSDT hot patching and this what is present in DSDT after the patch is applied :
`
        Method (_Q13, 0, NotSerialized)  // _Qxx: EC Query
        {
            If (ATKP)
            {
                \_SB.ATKD.IANE (0x32)
            }
        }

        Method (_Q14, 0, NotSerialized)  // _Qxx: EC Query
        {
            If (ATKP)
            {
                \_SB.ATKD.IANE (0x31)
            }
        }

        Method (_Q15, 0, NotSerialized)  // _Qxx: EC Query
        {
            If (ATKP)
            {
                \_SB.ATKD.IANE (0x30)
            }
        }
`
Also Used the AsusSMC debug kext to check for the _Q codes to confirm that the ones used in DSDT patch are corect and that the kext forwards the codes to OsX without any errors.

So I appreciate any hints on where to start looking for the issue 
Best regards.

hieplpvip, 2019-01-18:
Please send me your CLOVER folder

makito89, 2019-01-18:
I've attached the CLOVER folder including the original dump of ACPI (F4 in clover boot).
[CLOVER.zip](https://github.com/hieplpvip/AsusSMC/files/2774312/CLOVER.zip)

makito89, 2019-01-24:
After more test I think the issue is not with AsusSMC but with AppleALC.
I used the same CLOVER configuration (hot patches and kexts) with previous versions of OsX from 10.11 to 10.13 and it works as it should. The difference between them is the fact that in Mojave the sound is no longer obtained via clover patching of the AppleHDA kext but by using AppleALC kext.
Possible that the AppleALC kext reacts different to key codes ?

hieplpvip, 2019-09-15:
On Mojave, Apple removed a lot of layouts from AppleHDA, so traditional patching won't work. AppleALC is the only way to go now.

closed 2019-09-15 by hieplpvip