hieplpvip/AsusSMC #12: Patch DSDT giúp mình với bạn ơi

wanttolove, 2019-04-22:
[Send me macs-MacBook-Pro.zip](https://github.com/hieplpvip/AsusSMC/files/3104704/Send.me.macs-MacBook-Pro.zip)
Mình patch cái kbl của bạn vào thì báo lỗi WRAM is not accessible from this scope. Bấm fn+f3 thì vẫn nhảy popup bình thường nhưng ko có tác dụng gì cả. Thanks

hieplpvip, 2019-05-03:
Bạn dùng patch của Broadwell nha

hieplpvip, 2019-05-03:
Mình update patch rồi đó bạn. Bạn thử lại nha

wanttolove, 2019-05-04:
Lần này thì nó ko báo lỗi nữa nhưng vẫn như cũ bấm hiện ra nhưng ko tác dụng gì. windows vẫn sử dụng đc. đã thử cả 3 patch broadwell, kaby với cả whiskey. máy mình là kaby-R

hieplpvip, 2019-05-05:
Gửi mình file debug theo link này: https://www.tonymacx86.com/threads/tool-generate-proper-problem-reporting-files.235953/

wanttolove, 2019-05-05:
[debug_24314.zip](https://github.com/hieplpvip/AsusSMC/files/3145646/debug_24314.zip)
đây b ơi. thanks

hieplpvip, 2019-05-05:
[DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/3145691/DSDT.aml.zip)

Bạn xoá kext AsusNBFnKeys nha.

wanttolove, 2019-05-05:
ok được rồi thank bạn nhiều nhé. vn no1 :D

closed 2019-05-06 by hieplpvip