hieplpvip/AsusSMC #13: Backlight Keyboard WhiskeyLake Asus Notebook (Asus Vivobook UX333FN-A5811T)

andreszerocross, 2019-05-03:
I try this Kext and Patch for WhiskeyLake notebook,  with this code in SLKB
`Method (SLKB, 1, NotSerialized)
        {
            Store (And (Arg0, 0x7F), KBLV)
            If (And (Arg0, 0x80))
            {
                Store (DerefOf (Index (PWKB, KBLV)), Local0)
            }
            Else
            {
                Store (Zero, Local0)
            }

            ST9E (0x1F, 0xFF, Local0)
            Return (One)
        }`

So i can't set
`Method (SKBV, 1, NotSerialized)
            {
                Divide (Arg0, 0x10, , KBLV)
                ^^PCI0.LPCB.EC0.WRAM (0x09F0, KBLV)
                ^^PCI0.LPCB.EC0.ST9E (0x1F, 0xFF, Arg0)
                Return (Arg0)
            }`

I will get this error if i put this SKBV method. `Too few arguments, ^^PCI0.LPCB.EC0.WRAM required 3`

So i delete line "^^PCI0.LPCB.EC0.WRAM (0x09F0, KBLV)"

Backlight keyboard work but can't decrease and increase backlight. Just for information, this keyboard only has 1 button to increase and decrease Backlight keyboard (FN + F7).

I use Rehabman ACPIDebug.kext and Instrument EC,, i found the FN + F7 = method of _Q0D. I change Method _Q0D from "KBLU" to "KBLD",, after reboot when i repeat press FN + F7 ,, i saw decrease backlight in monitor, and backlight of keyboard is decreased too and turn off. But i can't increase backlight keyboard again (maybe because i change KBLU to KBLD in _Q0D). Then i change again "KBLD" to "KBLU" but after reboot,, Fn + F7 do nothing. No response, and keyboard backlight still in off status.

Any clue?? Please answer :)

hieplpvip, 2019-05-03:
@andreszerocross Please upload your DSDT

andreszerocross, 2019-05-03:
[Orignal DSDT + Patched DSDT.zip](https://github.com/hieplpvip/AsusSMC/files/3142462/Orignal.DSDT.%2B.Patched.DSDT.zip)

Here you go

hieplpvip, 2019-05-03:
How do you both increase/decrease keyboard backlight with just 1 button in Windows?

hieplpvip, 2019-05-03:
@andreszerocross Keep _Q0C and _Q0D as original. In method KBLU change 0xC7 to 0xC4.
Tell me if it works

andreszerocross, 2019-05-03:
I will try 0xC7 to 0xC4 and report here,, 

Btw, i don't make any change to _Q0C,, because when i use ACPIDebug.kext Fn + F7 = _Q0D. Maybe _Q0C for another combination button

andreszerocross, 2019-05-03:
In windows,, when you press it continue,, then it will be increase until high level (4 level), when you are at high level backlit, then if you press it again, it will turn off, and when you press again, it will start from level 1 backlit keyboard

andreszerocross, 2019-05-03:
@hieplpvip i change C7 to C4. It's work for increase backlight keyboard and there is notif in built-in display. But, no capability to decrease backlight

andreszerocross, 2019-05-03:
So, i have temporary solution.

When i track EC Query for FN + F6 with ACPIDebug.kext,, i found it related to _Q12. Then i made a change to _Q12 to

```
Method (_Q12, 0, NotSerialized)
        {
            KBLD ()
        }
```

And Increase and decrease backlight is working now,, normally FN + F6 is for disabling touchpad.

Yes i know this is temporary solution, maybe you have and idea to mix KBLD and KBLU in single Method Query

andreszerocross, 2019-05-03:
[image]
As you can see,, only single button to increase and decrease backlight keyboard as default

hieplpvip, 2019-05-04:
I haven't come up with a fix yet. However, there is a better temporary solution: Use Karabiner-Elements to remap F7 to decrease keyboard backlight. In this way, you don't have to sacrifice Fn+F6. Fn+F7 would increase keyboard backlight, and F7 only would decrease keyboard backlight.

yamvmax, 2019-06-28:
@andreszerocross 

I have same KB as you, could you please help me get kb lights working? I am fine with using f6, and f7 like you have.

andreszerocross, 2019-07-24:
> @andreszerocross
> 
> I have same KB as you, could you please help me get kb lights working? I am fine with using f6, and f7 like you have.

Upload your IOREG

yamvmax, 2019-07-24:
> > @andreszerocross
> > I have same KB as you, could you please help me get kb lights working? I am fine with using f6, and f7 like you have.
> 
> Upload your IOREG and your current DSDT.aml

It's here
https://github.com/hieplpvip/AsusSMC/issues/23#issuecomment-507436350