hieplpvip/AsusSMC #16: TODO

hieplpvip, 2019-05-07:
- [ ] Support backlight on USB keyboard. Ref: [Linux kernel](https://github.com/endlessm/linux/commit/aae7c7a42f03f030bb5cc2d5708200f1c571c063)

anonymous-writer, 2019-09-23:
Hi @hieplpvip , what do you meen with "Save ALS status across boot"?

hieplpvip, 2019-09-23:
@anonymous-writer You can press Fn+A to toggle ALS.

Ubsefor, 2019-10-17:
You can add `sudo mount -uw /` and `sudo mount -ur /` to mount system volume in rw and back in readonly, writing to /usr/bin generally requires root password anyway

hieplpvip, 2019-10-17:
I do not prefer it. Installing to /usr/local/bin would be enough

williambj1, 2019-12-15:
@hieplpvip 

Maybe this is useful https://github.com/acidanthera/VoodooInput/commit/3f6e60573b1317c6a3a18136cfbea9ee8eee6361

MohammadtaghiFarkhondekar, 2020-03-11:
I hope fix problem in 10.15...