hieplpvip/AsusSMC #18: Add Support For Skylake keyboard backlight

TineshSuresh, 2019-05-22:
Thanks For this driver. Actually most of the fn keys working except Keyboard backlight F3/F4.
The patches provided for Skylake keyboard is not matching in my dsdt.

I have attached the working patch for keyboard backlight:
I have to insert this code under this method :

Method SCDG {} :

            Name (BOFF, Zero)
            Method (SKBL, 1, NotSerialized)
            {
                If (Or (LEqual (Arg0, 0xED), LEqual (Arg0, 0xFD)))
                {
                    If (And (LEqual (Arg0, 0xED), LEqual (BOFF, 0xEA)))
                    {
                        Store (Zero, Local0)
                        Store (Arg0, BOFF)
                    }
                    ElseIf (And (LEqual (Arg0, 0xFD), LEqual (BOFF, 0xFA)))
                    {
                        Store (Zero, Local0)
                        Store (Arg0, BOFF)
                    }
                    Else
                    {
                        Return (BOFF)
                    }
                }
                ElseIf (Or (LEqual (Arg0, 0xEA), LEqual (Arg0, 0xFA)))
                {
                    Store (KBLV, Local0)
                    Store (Arg0, BOFF)
                }
                Else
                {
                    Store (Arg0, Local0)
                    Store (Arg0, KBLV)
                }

                Store (DerefOf (Index (KBPW, Local0)), Local1)
                ^^PCI0.LPCB.EC0.WRAM (0x04B1, Local1)
                Return (Local0)
            }

            Name (KBPW, Buffer (0x10)
            {
                /* 0000 */  0x00, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77,
                /* 0008 */  0x88, 0x99, 0xAA, 0xBB, 0xCC, 0xDD, 0xEE, 0xFF 
            })
            Method (GKBL, 1, NotSerialized)
            {
                If (LEqual (Arg0, 0xFF))
                {
                    Return (BOFF)
                }

                Return (KBLV)
            }
        }

The keyboard backlight fn keys working with elmydinesh kext but not urs
Hope this helps you
Thank you

hieplpvip, 2019-05-22:
My current patch for Skylake is similar to the patch working for you. There's no way to know why without your EFI folder.

Please send me your EFI folder

TineshSuresh, 2019-05-22:
[EFI.zip](https://github.com/hieplpvip/AsusSMC/files/3207159/EFI.zip)

mishurov, 2020-03-28:
@TineshSuresh critical line here is ```^^PCI0.LPCB.EC0.WRAM (0x04B1, Local1)```, @hieplpvip uses an ACPI method only for setting the value for keyboard light, for everything else he uses NVRAM, so ```kbl_broadwell.txt``` should work, although my dsdt have no ```KBLD``` and ```KBLU``` methods, I only inserted ```SKBV```. Note, it has 16 brightness levels, you need to press buttons more times to notice changes.

closed 2020-09-15 by hieplpvip