hieplpvip/AsusSMC #20: Mac os 10.15beta Catalina

YUWEN9410, 2019-06-10:
The keyboard backlight failed in 10.15 beta.

YUWEN9410, 2019-06-10:
asus N552vw6700

anonymous-writer, 2019-06-15:
Also on ASUS Zenbook UX430UAK. The brightness will not changed but the icon appears.

I would also like to thank you for your previous work. Until Mojave the kext works perfectly.

hieplpvip, 2019-06-15:
#### About keyboard backlight:
Have you updated VirtualSMC to latest version?
Keyboard backlight depends on SMC. That keyboard backlight doesn't work while the other functions work suggest that there's a problem with SMC simulation.

#### About screen brightness:
AsusSMC only sends fake keyboard press event (like on real Mac) to make macOS change brightness. It doesn't change brightness directly.

The fact that the badge appears means that system has successfully receive the fake keyboard event. Brightness still not working suggests that the brightness patch is not working properly.

You can try setting brightness manually in System Preferences. If it doesn't work, the problem is not with AsusSMC.

anonymous-writer, 2019-06-15:
Sorry @hieplpvip ,
I write brightness and mean keyboard backlight. Screen Brightness work perfekt.

I use VirtualSMC  1.1.1.

hieplpvip, 2019-06-15:
I'd wait for final release of Catalina before updating. There's not much I can help you now, sorry.

anonymous-writer, 2019-06-15:
Small temporary solution meanwhile, this works great with Catalina and AsusSMC and keyboard backlight .

https://www.heise.de/download/product/lab-tick-37046/download

YUWEN9410, 2019-06-16:
Thanks to the author @hieplpvip  for his efforts, I have tried to use AsusNBFnKeys.kext. The keyboard backlight is working properly, but the keyboard backlight icon does not appear on the keyboard backlight because the drive is too old.

YUWEN9410, 2019-06-16:
I am now rolling back to mojave, and the keyboard backlight is normal. Maybe catalina has some bugs

hieplpvip, 2019-06-16:
@anonymous-writer What is that?

anonymous-writer, 2019-06-16:
Hello @hieplpvip ,
A keyboard backlight slider that works great with your kext and that also already works under Catalina.

[image]

MohammadtaghiFarkhondekar, 2019-09-15:
??? please make kext for 10.15

hieplpvip, 2019-10-10:
Hi everyone.

I've just upgraded to Catalina and had some look at the problem. It seems that macOS Catalina change how it detects if keyboard backlight is supported. On Mojave and below, simulating 2 SMC keys (LKSB and LKSS) is enough. However, that no longer works. In Keyboard section in System Preferences, we no longer see anything related to keyboard backlight (which, obviously, show that Catalina thinks that keyboard backlight is not supported).

The reason why LabTick works (I guess) is that it interfere directly with those 2 SMC values, and ignore the check for keyboard backlight existence. For now, you guys should use it.

Fixing this problem would take time. I'm still not sure how to fix it...

anonymous-writer, 2019-10-11:
Thank you for your feedback. I will also try to find out how the keyboard backlight are addressed at Catalina. Did I find something I write you.

anonymous-writer, 2019-10-12:
Where in the source code can you customize the output keys? You could temporarily spend another key output. To write out temporarily a different short command for use with LabTick.
[image]

anonymous-writer, 2019-10-12:
This is also interesting and changes the keyboard brightness.

[https://github.com/pirate/mac-keyboard-brightness](url)

hieplpvip, 2019-10-13:
Let me explain how each program works.

AsusSMC works by simulating SMC keys (namely LKSB, LKSS, etc. you can find them in the source code).. When macOS (Mojave and below) detects these keys, it will think that keyboard backlight is supported and show the settings for keyboard backlight. AsusSMC also simulates an Apple keyboard, so everytime a user presses Fn keys, it will send a fake event to macOS. macOS then changes the value of LKSB, which will call the update function in AsusSMC (in KeyImplementation.cpp), which will call ACPI methods to actually set the keyboard backlight value.

The methods where macOS set the value of LKSB lie in AppleLMUController.kext (which is renamed to AppleSMCLMU.kext on Catalina). Labtick/macbrightness use these method.

I believe the problem on Catalina is that it adds some more check, and I'm pretty sure it's not in AppleSMCLMU (I have tried RE it but found nothing interesting). I will RE more when I have time.

Actually, a workaround for Catalina is pretty simple. When Fn keys are pressed, we'll call ACPI methods directly, but in that way we'll lose some nice feature of macOS (auto turn off, ...), so I don't prefer it. If you guys want, I may implement it before I found a proper solution.

gulios, 2019-10-13:
IMO without auto-off is much better than LabTick ;) so if you can ;)

hieplpvip, 2019-10-13:
I'll implement it and release a new version next week.

hieplpvip, 2019-10-13:
[AsusSMC-1.2.0-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/3722272/AsusSMC-1.2.0-RELEASE.zip)

I have added workaround for Catalina. Please test it. Don't forget to install AsusSMCDaemon. Without it you won't see the badge.

Technical explanation: Since I can't get macOS to show the badge, I have to show it manually via AsusSMCDaemon. This has two drawbacks: you have to install it and it will not work at login window (macOS doesn't load daemons there).

gulios, 2019-10-13:
it works ;))

anonymous-writer, 2019-10-13:
Me too. You are the best.

MohammadtaghiFarkhondekar, 2019-10-13:
Is not working for me ! And after using kext my keyboard doesn't working like old v in 10.15

hieplpvip, 2019-10-14:
@MohammadtaghiFarkhondekar Are you still using Mojave? If yes, please test it on Mojave. It doesn't support HID on Catalina

SkreamDark, 2019-10-14:
Hello, thank you for new kext) now my backlight keyboard working, but not like on Mojave. In Mojave keyboardbacklight can be turn off automatically after 10 second or more. How Can i return this option?
In Mojave
[image]
In Catalina
[image]

gulios, 2019-10-14:
> Hello, thank you for new kext) now my backlight keyboard working, but not like on Mojave. In Mojave keyboardbacklight can be turn off automatically after 10 second or more. How Can i return this option?
> In Mojave
> [image]
> In Catalina
> [image]

You didn't read this issue, right ? ;)

SkreamDark, 2019-10-14:
Just very quick), but i understand, that it may be just incompatible with Catalina right now
When i saw new kext, i was so glad)
I hope its will be fixed very soon

anonymous-writer, 2019-10-14:
If you would read what is written above then all questions would be answered.

In any case, I thank the developer that he work on this kext. He finally gets nothing for this work except somtime a donation.

MohammadtaghiFarkhondekar, 2019-10-14:
I test again in 10.14.6 all off fn keys work good but keyboard back light fn key aren't working ! !!!

hieplpvip, 2019-10-15:
@MohammadtaghiFarkhondekar Test this one on Mojave
[1.2.0.RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/3726961/1.2.0.RELEASE.zip)

MohammadtaghiFarkhondekar, 2019-10-15:
No , keyboard back light FN keys aren't working again.

hieplpvip, 2019-10-16:
Could someone still having Mojave installed please send me a copy of CoreBrightness?
The location of it is `/System/Library/PrivateFrameworks/CoreBrightness.framework/Versions/A/CoreBrightness`

SkreamDark, 2019-10-16:
Here, i hope its will help you and thanks for you work) 
[CoreBrightness.framework.zip](https://github.com/hieplpvip/AsusSMC/files/3736046/CoreBrightness.framework.zip)

hieplpvip, 2019-10-17:
Does anyone have a real MacbookPro running Catalina? I need IOReg of it.

Thanks to Apple who rewrote CoreBrightness completely.

andrewwyatt, 2019-10-17:
> Does anyone have a real MacbookPro running Catalina? I need IOReg of it.

Yep...2014 MBP

[ioreg.zip](https://github.com/hieplpvip/AsusSMC/files/3739294/ioreg.zip)

MohammadtaghiFarkhondekar, 2019-10-18:
[CoreBrightness.framework.zip](https://github.com/hieplpvip/AsusSMC/files/3743649/CoreBrightness.framework.zip)

I'm using the new Releases in 10.14.6 and 10.15 but  ,,,  asus aura back light is not working in macos 10.14.6 and in 10.15 my keyboard is not working ( any key is not working !!!)

hieplpvip, 2019-10-18:
@andrewwyatt Can you dump ioreg again, this time using [IORegistryExplorer](https://raw.githubusercontent.com/black-dragon74/OSX-Debug/master/IORegistryExplorer.zip)?

andrewwyatt, 2019-10-18:
> @andrewwyatt Can you dump ioreg again, this time using [IORegistryExplorer](https://raw.githubusercontent.com/black-dragon74/OSX-Debug/master/IORegistryExplorer.zip)?

I sent it direct, don't want to post it since I can't scrub my serials.

SkreamDark, 2019-10-18:
i have MacBook Air, will his ioreg be useful??

ertanyanardag, 2019-10-19:
Thank you, it's great work...

DilVika, 2020-02-08:
Have this problem with Keyboard backlight was fixed by the newest version, anyone? I have updated to newest ver but Keyboard backlight just not work at all.

Ubsefor, 2020-02-08:
> Have this problem with Keyboard backlight was fixed by the newest version, anyone? I have updated to newest ver but Keyboard backlight just not work at all. My laptop is Gl552vx and everything worked fine on Mojave.

Have you updated Lilu and VirtualSMC?
For me everything works perfectly

MohammadtaghiFarkhondekar, 2020-03-08:
up..... need fixing .

valinor61, 2020-03-21:
I am getting some progress on this issue finally i manage to work that kext in catalina but its not a solution for now. Please dont try for your daily usage it may break your system.
- I compared ioregs in catalina and mojave the key difference is a driver extension called AppleUserHIDDrivers.dext in catalina.
- In mojave there is no such driver so we can use asussmc with IOHidEventDriver but in catalina we can't use that driver because AppleUserHIDDrivers.dext interfere and connect asussmc because its a default hid behaviour on catalina. 
- I removed AppleUserHIDDrivers.dext its under System/Library/Driver Extensions. After lots of lagging and hid issues asussmc reconnected IOHidEventDriver instead of AppleUserHIDDriver and asussmc worked for a while.
- There are two way to fixing it first one is fixing incompatilibity on AppleUserHIDDrivers and AsusSmc(permenant solution), second block AppleUserHidDrivers for attached keyboard.

williambj1, 2020-03-22:
@valinor61 

Nice research!
I think we need to beat this dext from loading under the keyboard with a higher `IOProbeScore` in AsusSMC.

[image]

valinor61, 2020-03-22:
@williambj1 thanks IOProbeScore override actualy did what I said but i think still AsusSmc connecting with AppleUserHIDDrivers.dext still no keyboard input. I attached text cases and my IOProbe Override info.plist kext.
[AsusSmcCatalina.kext.zip](https://github.com/hieplpvip/AsusSMC/files/4364981/AsusSmcCatalina.kext.zip)

[image]
 Catalina - No kext installed 
[image]
 Mojave - With AsusSmc
[image]
 Catalina - IOProbeScore Override
[image]
 Catalina - AsusSmc and IOProbeScore Override

closed 2020-08-17 by hieplpvip

gulios, 2020-08-18:
Cool, works on BS too;)

Ubsefor, 2020-08-18:
> Cool, works on BS too;)

Could you please post your EFI folder? I’m having troubles getting AsusSMC to work under Clover 5120 Catalina

gulios, 2020-08-18:
i don't use clover anymore

Ubsefor, 2020-08-18:
> i don't use clover anymore

Then could you please check, whether AsusSMCDaemon is loaded?
(use sudo launchctl list | grep Asus )

rafaelmaeuer, 2020-12-27:
> Cool, works on BS too;)

What does work for you in BS? Backlight in general or changing backlight brightness with fn-keys?

gulios, 2020-12-27:
> > Cool, works on BS too;)
> 
> What does work for you in BS? Backlight in general or changing backlight brightness with fn-keys?

Backlight:
- fn keys
- "turn keyboard backlight off after.." from keyboard settings
- fn+A - but it works odd so i don't use it

gulios, 2020-12-27:
> > i don't use clover anymore
> 
> Then could you please check, whether AsusSMCDaemon is loaded?
> (use sudo launchctl list | grep Asus )

yes it's loaded
```
[17:24:22] gulios :: GuliossBookHack  ➜  ~ » launchctl list | grep Asus
500	0	com.hieplpvip.AsusSMCDaemon
```

rafaelmaeuer, 2020-12-28:
> > > Cool, works on BS too;)
> > 
> > 
> > What does work for you in BS? Backlight in general or changing backlight brightness with fn-keys?
> 
> Backlight:
> 
> * fn keys
> * "turn keyboard backlight off after.." from keyboard settings
> * fn+A - but it works odd so i don't use it

strange for me its not working, linked my issue

shiecldk, 2021-08-12:
Could anyone confirm if this still work in Catalina 10.15.7 and Big Sur?

rafaelmaeuer, 2021-08-12:
Yes I can confirm its working on macOS Big Sur 11.4 with OpenCore 0.7.1, check [rafaelmaeuer/Asus-UX32VD-Hackintosh](https://github.com/rafaelmaeuer/Asus-UX32VD-Hackintosh) for details.

shiecldk, 2021-08-14:
@rafaelmaeuer Could  you share your SSDT or EFI folder you use for it?