hieplpvip/AsusSMC #21: Fn key mapping - KeyboardLight not working and hibernate not working

bonaw, 2019-06-12:
Can you help me?, i tried anything but nothing works 
I have a core i3-7020U Asus vivobook 14 X420

PD: There you have my efi folder
https://inacapmailcl-my.sharepoint.com/:u:/g/personal/felipe_gonzalez246_inacapmail_cl/EZXOpf1tdVFFkzZory3QMj8Bfnlysw243QKIfrcgnnWnjQ?e=U9jPVJ

hieplpvip, 2019-06-12:
You're using FakeSMC, not VirtualSMC

bonaw, 2019-06-15:
All working now!, thanks for your help.
Can you help me to get my fn layout? it's different for my X420.

PD: Do you have any idea of why I can`t get hibernation working?

bonaw, 2019-06-15:
Any idea of why it takes forever for my pc to boot with VirtualSMC?

hieplpvip, 2019-06-16:
I'm not sure if Hibernation would ever work on Hackintosh.

Did you remove SMCHelper?

fgvilches, 2019-06-16:
I removed SMCHelper but still taking forever to boot.
Also, can you help me with de fn key mapping for my model?.
by hibernation I mean Sleep/Wake Function... 
Some videos on youtube show that works.

hieplpvip, 2019-06-16:
I don't have time to look thoroughly your problem.

There's already a full working build for Vivobook that can help you [here](https://github.com/tctien342/Asus-Vivobook-S510UA-High-Sierra-10.13-Hackintosh)

closed 2019-06-18 by fgvilches