hieplpvip/AsusSMC #23: Backlight Keyboard  ASUS VivoBook Flip 14 TP412UA-IH31T Core i3-8130U

yamvmax, 2019-06-28:
Cant get KB light to turn on. Using asusSMC, and fn patches, and KB patch. My keyboard only has 1 fn key for KB light. Please help.

hieplpvip, 2019-06-29:
Attach your DSDT

yamvmax, 2019-06-29:
[DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/3341279/DSDT.aml.zip)

yamvmax, 2019-06-29:
Here it is

yamvmax, 2019-06-30:
Should I use asusfn keys???

hieplpvip, 2019-07-01:
@yamvmax I saw no method relating keyboard backlight in your DSDT.
Please attach your IOReg report.

yamvmax, 2019-07-01:
@hieplpvip will do later on. Thanks. BTW, I believe I have the same kb as someone else here posted. Just one fn key for the kb light.

yamvmax, 2019-07-01:
Send me richs-MBP.zip

Here you go, just unzip. Everything is in there. Thanks

yamvmax, 2019-07-01:
[Send me richs-MBP.zip](https://github.com/hieplpvip/AsusSMC/files/3347419/Send.me.richs-MBP.zip)

yamvmax, 2019-07-01:
sorry, there it is

hieplpvip, 2019-07-02:
@yamvmax Your laptop uses a new method for changing keyboard backlight.
I'm working on support for HID-based keyboard. Will add support for you when I finish

yamvmax, 2019-07-02:
Ok, great! Thanks.

yamvmax, 2019-07-02:
Any possibility of some workaround in the meantime? If not, I'm willing to test anything you come up with! Thanks again.

hieplpvip, 2019-07-02:
Use `ACPIDebug` to find which method is called in DSDT when pressing Fn keys for keyboard backlight. (I believe it is `_Q66`, but I need confirmation)

yamvmax, 2019-07-03:
Yes, i am getting Q66