hieplpvip/AsusSMC #27: GL553VD

MohammadtaghiFarkhondekar, 2019-07-17:
HI.
please add Asus Aura core for the laptop to your kexts bc aura keyboard backlight doesn't work.

MohammadtaghiFarkhondekar, 2019-07-17:
https://github.com/wroberts/rogauracore/issues/4
https://github.com/wroberts/rogauracore

hieplpvip, 2019-07-17:
Test this one please [1.2.0.RELEASE.zip](https://files.gitter.im/hieplpvip/AsusSMC/vvPz/1.2.0.RELEASE.zip)

MohammadtaghiFarkhondekar, 2019-07-17:
Thank you . Does it need to patching for working ?

MohammadtaghiFarkhondekar, 2019-07-17:
Ok . I test it without any patching .
F1+FN working good
F2+FN I think working bc I haven't wifi card for set it on turn off mode.
F3 and F4 aren't working 
F5 and F6 are working so good but F14 and F15 alsoe are working
F7 is working 
F8 I think it doesn't working .
F9 is working but without showing the system notification for turn off track pad.
F10,F11,F12 are working good.

hieplpvip, 2019-07-17:
On Windows/Linux, Aura driver sends commands to the keyboard through the USB protocol.
Unfortunately, I've not figured out how to to it properly on macOS, so keyboard backlight is not working now.

MohammadtaghiFarkhondekar, 2019-07-17:
That's right . Aura core has open source project in GitHub I think it can help for fixing .

MohammadtaghiFarkhondekar, 2019-07-17:
https://www.asus.com/campaign/aura/fl/Partners-and-promotions.html

hieplpvip, 2019-07-17:
I've already known all you said

MohammadtaghiFarkhondekar, 2019-07-17:
https://gitlab.com/thelastguardian/openpyaura

hieplpvip, 2019-07-18:
[rogauracore.zip](https://github.com/hieplpvip/AsusSMC/files/3404755/rogauracore.zip)

Hi @MohammadtaghiFarkhondekar, can you test this? It's compiled for macOS from source code of rogauracore that you gave

MohammadtaghiFarkhondekar, 2019-07-18:
Hi. Thank you so much. how can I test it?

hieplpvip, 2019-07-18:
Another one has tested it and confirmed it doesn't work...

MohammadtaghiFarkhondekar, 2019-07-18:
I tested ubuntu 18.04 and it have Aura driver and it can control keyboard backlight.

hieplpvip, 2019-07-18:
Test this [one](https://files.gitter.im/hieplpvip/AsusSMC/O3yC/1.2.0.RELEASE.zip)

Note that macOS supports 16 levels, while your keyboard only support 4 level, so:
- Level 0-4 of macOS -> level 0 of your keyboard
- Level 5-8 of macOS -> level 1 of your keyboard
- Level 9-12 of macOS -> level 2 of your keyboard
- Level 13-16 of macOS -> level 3 of your keyboard

MohammadtaghiFarkhondekar, 2019-07-18:
Thank you. just replace kext?
I replaced kext and reboot again but keybord backlight with FN key (f3,f4) doesn't work.

hieplpvip, 2019-07-18:
Where do you install kexts?
If it is /L/E, have you rebuilt kext cache?

MohammadtaghiFarkhondekar, 2019-07-18:
Clover folder . Need to install in the system ?

hieplpvip, 2019-07-18:
CLOVER should be good.
Please send me your IOReg

MohammadtaghiFarkhondekar, 2019-07-18:
Ok. I'm going to get file now and send it

MohammadtaghiFarkhondekar, 2019-07-18:
[MacBook Pro-rog553vd.zip](https://github.com/hieplpvip/AsusSMC/files/3407306/MacBook.Pro-rog553vd.zip)

hieplpvip, 2019-07-18:
@MohammadtaghiFarkhondekar Does the badge show up when you press Fn+F3/F4?

MohammadtaghiFarkhondekar, 2019-07-18:
Hi. No. Any things .

williambj1, 2019-07-19:
> Hi. Thank you so much. how can I test it?
> I opened it =>
> [image]

Seems that you're missing libusb, :)
`brew install libusb` should solve that error.
However, I've confirmed that this didn't work for me. Some people say that libusb is dead on macOS Mojave, currently, I don't have HS.

hieplpvip, 2019-07-19:
@MohammadtaghiFarkhondekar [1.2.0.RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/3409591/1.2.0.RELEASE.zip)

MohammadtaghiFarkhondekar, 2019-07-19:
Oh my God . Woooooooooooow . Thank you so much . Your work is so nice. 
Now keyboard backlight working great but without  showing badge like touchpad (F9).
Can you make manager app for Aura core like Linux project ? 
Thank you

hieplpvip, 2019-07-19:
@MohammadtaghiFarkhondekar will do when I have time, not now sorry

closed 2019-07-19 by hieplpvip

MohammadtaghiFarkhondekar, 2019-07-19:
Thank you, your project link add to [https://github.com/MohammadtaghiFarkhondekar/macOS-Mojave-For-Asus-ROG-GL553VD/blob/master/README.md](url)

hieplpvip, 2019-07-20:
[ROGAuraCore.zip](https://github.com/hieplpvip/AsusSMC/files/3413504/ROGAuraCore.zip)
@MohammadtaghiFarkhondekar test this

MohammadtaghiFarkhondekar, 2019-07-20:
Hi.
[image]
 thank you. how can I use it?

MohammadtaghiFarkhondekar, 2019-07-20:
woooooorking

MohammadtaghiFarkhondekar, 2019-07-20:
[image]
can you make UI for it?

MohammadtaghiFarkhondekar, 2019-07-20:
wooooooooooooooooow . its wooooooooork soooooo goooooooooooooood . 
it's need UI . it's better than asus app for aura !
it's the first app for aura for mac on the word by you!

hieplpvip, 2019-07-20:
@MohammadtaghiFarkhondekar I've published the program at [macrogaura](https://github.com/hieplpvip/macrogaura/)

I'm not good at designing UI. You'd better wait for someone else to do it.

MohammadtaghiFarkhondekar, 2019-07-20:
it's a good app. thank you for making it. 
I think it doesn't need pro UI  .
Asus aura core app in windows 
[image]

MohammadtaghiFarkhondekar, 2019-07-20:
I add your project to gl553vd page and I will make a video about it .