hieplpvip/AsusSMC #28: Can't find 'ALS' in the 'DSDT'

angeljavan, 2019-07-29:
My noteBook is FX80GE(FX504GE),I can't find 'ALS' in the 'DSDT',but I cant find 'ALSC' and 'ALSS'.when  apply patch [als] Patch ALSS ,There was a mistake in the 'DSDT'

williambj1, 2019-07-30:
ROG Keyboards have different key caps :)

angeljavan, 2019-07-31:
Can this product only be used in the ROG series?

angeljavan, 2019-07-31:
These two files will conflict ，the voodooI2c ,the ASUSSMC

williambj1, 2019-08-01:
I suppose it's not an issue caused by VoodooI2C, I have the same keycaps as yours (ROG) and all Fn keys work well on the correct keycaps.

- Chinese Translation
>
> 我觉得应该不是 VoodooI2C 的问题，我的键帽和你的一样，都是 ROG，而且所有的 Fn 键都正常工作在正确的键位上

angeljavan, 2019-08-02:
You are right. But my touchpad feels a bit stuttering, what is the problem?