hieplpvip/AsusSMC #35: Dependency com.apple.iokit.IOHIDSystem was not found for kext com.hieplpvip.AsusSMC

williambj1, 2019-09-13:
Hi! @hieplpvip 

When I was debugging with OpenCore Logs, I found this

```log
00:000 00:000 Dependency com.apple.iokit.IOHIDSystem was not found for kext com.hieplpvip.AsusSMC
00:000 00:000 Dependency com.apple.iokit.IOHIDSystem fallback to com.apple.iokit.IOHIDSystem succeeded. Please fix your kext!
```

hieplpvip, 2019-09-15:
I'll look into it after I solve the problem with HID-based keyboard. I also prefer to wait for stable version of OpenCore.

iNDi, 2019-10-02:
from the OpenCore docs
  //
  // Some kexts, notably VoodooPS2 forks, link against IOHIDSystem.kext, which is a plist-only
  // dummy, macOS does not add to the prelinkedkernel. This cannot succeed as /S/L/E directory
  // is not accessible (and can be encrypted). Normally kext's Info.plist is to be fixed, but
  // we also put a hack here to let some common kexts work.
  //
so change
com.apple.iokit.IOHIDSystem to com.apple.iokit.IOHIDFamily
in AsusSMC.kext/Contents/Info.plist

hieplpvip, 2019-10-06:
Fixed. Thanks.

closed 2019-10-06 by hieplpvip

salim0786, 2020-08-01:
Hey @hiepvip, could you please guide me where to change it. I mean I'll probably have to add some keys, right? Sorry I'm a noob at this.