hieplpvip/AsusSMC #36: Asus VivoBook S530Fn - Fn Keys are different

tvieira20, 2019-09-23:
Hi. I managed to fix the keyboard backlight with AsusSMC and your patch - which I must say thank you for making it possible.
I have fully functioning Keyboard Light + 16 Light intensity after patching "[kbl] whiskeylake".
System is Asus VivoBook S15 S530Fn > i7-8565U, 8GB RAM, GeForce MX150 (disabled).

Although, there are some issues with the Fn Keys (I used Karabiner to remap the keys temporarily) as they are different on my laptop:

F1/F2/F3: Mute/Decrease/Increase volume
F4/F5: Decrease/Increase screen backlight
F6: Enable/Disable trackpad
F7: Keyboard Backlight  (3 stages in Windows 10)
F8: Switch monitor (Unchanged)
F9: Home
F10: End
F11/F12: Page UP/Down

I remaped with Karabiner F6/F7 to "Decrease/Increase keyboard backlight" because I don't use Disable Trackpad at all.
Do you need me to upload DSDT so yo ucan see what I have going on?

Thanks

[image]

hieplpvip, 2019-09-23:
Well, if you just folllowed the instruction and it works, no need to update the DSDT

tvieira20, 2019-09-23:
> Well, if you just folllowed the instruction and it works, no need to update the DSDT

Nevermind, I didn't understand what I needed to change in the code in order to re-organize the keys. 
Thanks again for all your work!

closed 2019-09-23 by tvieira20