hieplpvip/AsusSMC #45: Asus GL503GE Keyboard FN Keys Not working completely

drg619, 2019-10-29:
Sir
with asussmc.kext only FN + F1(Mute) , F2(Previous) , F3(Play/Pause) , F4(Next) are working rest of the keys like disable touchpad, screen off.etc.....
also keyboard backlight controls  using FN+Up/Down Arrow keys not working.

i tried applying your patches to dsdt but im geting compile error when i patch my cpu.

plz can u help me with the issue?

drg619, 2019-10-30:
[DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/3789150/DSDT.aml.zip)
here is my dsdt file plz help

black-dragon74, 2019-11-02:
Your device uses a HID based keyboard. The patches are for WMI based.

drg619, 2019-11-03:
> Your device uses a HID based keyboard. The patches are for WMI based.

How do i fix it?