hieplpvip/AsusSMC #46: Keyboard backlight remains off after sleep

cowpod, 2019-10-30:
this is not a big issue, but if i set the brightness to max, close the lid, and then wake it, the backlight remains off until the brightness is changed again.
ASUS S410U/X411U

LeeBinder, 2019-12-21:
in Catalina, Mojave, or in both?

LeeBinder, 2020-01-05:
@cowpod Henry? see my inquiry to you above.

cowpod, 2020-01-05:
> @cowpod Henry? see my inquiry to you above.

Only in Catalina. Worked fine in Mojave for me too.

AR-CADE, 2020-02-03:
Can someone test this version, I added a workaround for the backlight after sleep, by implementing PM events as described [here](https://developer.apple.com/library/archive/documentation/DeviceDrivers/Conceptual/IOKitFundamentals/PowerMgmt/PowerMgmt.html)
[AsusSMC-1.2.1-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/4146648/AsusSMC-1.2.1-RELEASE.zip)

LeeBinder, 2020-02-03:
Thank you @AR-CADE , working now as supposed in Catalina. Problem is that now backlight does not work at anymore in Mojave, which was fine before with hieplpvip's original (not tested in High Sierra yet but most likely it's the same there).

That can be for two possible reasons:
1. you only have Catalina SDK installed in Xcode so the kext won't generally work in <10.15. If so, can you please also install Mojave and High Sierra SDKs and include these macOS versions in compile options
2. if you do have Mojave and High Sierra SDKs installed and compiled for these OSes, too, can you please introduce a function which only triggers PM events like this:
```
if macOS version = 10.15 then
  PM events
end if
```
Thanks for your efforts.

AR-CADE, 2020-02-03:
Hi @LeeBinder , 

1) you are right , i only have 10.15 SDK installed, that's surely why the provided binaries don't work on  pre-catalina

can you checkout and build my fork ([link here](https://github.com/AR-CADE/AsusSMC)) with the desired SDK to see if it work 

2) the pseudo-code that you advise is already implemented, so this fix should not affect older Mac OS version :-)

I have also requested a "pull request" , see [this link](https://github.com/hieplpvip/AsusSMC/pull/62/files) for differences

LeeBinder, 2020-02-03:
@AR-CADE great. I totally can't provide any time right now to spend in Xcode, so either we'll just wait for hieplpvip to build the kext incl. your pull request, or you simply add Mojave and High Sierra SDK for compatibility, as has hieplpvip. Be aware that many h'toshers still use Mojave.

bliss95, 2020-02-04:
@AR-CADE Your fix works properly for Asus UX430UNR backlight.
Thank you.

closed 2020-02-06 by hieplpvip

LeeBinder, 2020-02-06:
confirmed solved with latest 1.2.2 - thank you hieplpvip and AR-CADE!