hieplpvip/AsusSMC #49: HID Keyboard doesn't work on macOS Catalina

black-dragon74, 2019-11-02:
Hey there,

I have an ASUS ROG GL-503GE laptop. The keyboard is HID based on USB bus.

When I use AsusSMC the keyboard does not work at all. I have attached debug log and also the IOREG.

Please take a look.

Regards
[Debug.zip](https://github.com/hieplpvip/AsusSMC/files/3800945/Debug.zip)

hieplpvip, 2019-12-22:
Hi,

This kext works on some ROG laptop, so I don't really know why it doesn't work for you guy. Currently I don't have much time for this kext, so I can't promise when I can fix this.

I'm also focusing more on bring full support for Catalina (Apple rewrote their kexts, so the old method doesn't work anymore).

If anyone figure out how to fix this, I'll be very happy to merge that fix.

valinor61, 2019-12-22:
in my reg and black-dragons reg as well it seems both asussmc and keyboard loads correctly I ve also tried debug version and debug boot arg I still didnt get any error. i am guessing somehow asussmc overrides keyboard layout because in mojave normally without asussmc my " character works correct but when I install asus smc it's become < character. Is there any way to disable that override. That might fix our problem.

hieplpvip, 2019-12-22:
@valinor61 Try deleting methods `handleInterruptReport` and `dispatchKeyboardEvent` in AsusHIDDriver (both `cpp` and `hpp`)

valinor61, 2020-01-15:
@hieplpvip sorry for long delay I had exams. I ve analyzed AsusHIDDriver.cpp  i find the origin of the error but I have no idea how to fix it. 
super::start(provider) function in AsusHIDDriver.cpp seems the issue. If that function starts application fails. Its not even printing "Error loading HID driver" warning also not returning false. It seems there is a difference in catalina in that code somehow. 
Kext simply crashing so thats why we cant use kb. If I only delete super::start(provider) I can use my keyboard but kext is not working that way apperently.

cowpod, 2020-01-28:
I don't know how to undo emoji reactions.
Keyboard doesn't work with AsusSMC unless return true is added, where as @valinor61 wrote it doesn't do anything.
ASUS ROG GL504GW
[log.txt](https://github.com/hieplpvip/AsusSMC/files/4120288/log.txt)

andreszerocross, 2020-01-30:
> @valinor61 Try deleting methods `handleInterruptReport` and `dispatchKeyboardEvent` in `AsusHIDDriver` (both `cpp` and `hpp`).
> 
> If keyboard still doesn't work, try disabling `AsusHIDDriver` by leaving only `return true;` in method `start`.

Does AsusSMC.kext r1.2.1 fix this HIDDevice?? Most ASUS ROG laptop with HID device cant use internal keyboard when using AsusSMC.kext

cowpod, 2020-01-30:
> Does AsusSMC.kext r1.2.1 fix this HIDDevice?? Most ASUS ROG laptop with HID device cant use internal keyboard when using AsusSMC.kext

still broken, catalina 10.15.2 1.2.1

andreszerocross, 2020-02-06:
> > > @valinor61 Try deleting methods `handleInterruptReport` and `dispatchKeyboardEvent` in `AsusHIDDriver` (both `cpp` and `hpp`).
> > > If keyboard still doesn't work, try disabling `AsusHIDDriver` by leaving only `return true;` in method `start`.
> > 
> > 
> > Does AsusSMC.kext r1.2.1 fix this HIDDevice?? Most ASUS ROG laptop with HID device cant use internal keyboard when using AsusSMC.kext
> 
> still broken, catalina 10.15.2 1.2.1
> edit: same on 10.15.3

Yes, i just tested it :(

valinor61, 2020-03-05:
I find something odd that might help fixing the issue .With karabiner app I can use AsusSMC and my keyboard but all fn keys are not working in that way only print key is working. Normally volume controller keys were working.

valinor61, 2020-03-12:
@hieplpvip Sorry to bother you but i think i find something. 
Left picture is asussmc kext installed.
Right one is not installed.

- In right part AsusSmc calls -> VirtualHidKeyboard -> AppleUserHidEventDriver
NKey devices -> AppleUsbHostHidDevice -> AsusHidDriver
- But in left Asus NKey devices -> AppleUsbHostHidDevice -> AppleUserHidEventDriver
It explains all wrong behavior on hid devices.
In order the keyboard work nkey devices must call to AppleUserHidEventDriver in AsusHidDriver no regular key identified so we cant use keyboard.
In order to Asussmc works it has to call AsusHidDriver in AppleUserHidEventDriver there is nothing for AsusSmc thats why fn keys not work either. If am I right you can fix it easily.
[image]

williambj1, 2020-03-22:
https://github.com/hieplpvip/AsusSMC/issues/20#issuecomment-602153129

This might explain why. I'm totally new in IOKit, just did some research recently, sorry if I posted garbage.

williambj1, 2020-03-22:
Hmmm, this is interesting.
After adding `IOProbeScore` under

https://github.com/hieplpvip/AsusSMC/blob/3991e322a8e12752c311df73ddc4cdaaeab7e1b7/Info.plist#L96-L97

```patch
diff --git a/Info.plist b/Info.plist
index 8356b54..882021d 100644
--- a/Info.plist
+++ b/Info.plist
@@ -95,6 +95,8 @@
                        <string>AsusHID</string>
                        <key>IOClass</key>
                        <string>AsusHIDDriver</string>
+                       <key>IOProbeScore</key>
+                       <string>5000</string>
                        <key>IOProviderClass</key>
                        <string>IOHIDInterface</string>
                        <key>ProductID</key>
```

it gives me a kernel panic, still good news since it's at least loaded.

```log
panic(cpu 8 caller 0xffffff80200652fa): Kernel trap at 0xffffff7fa3211ccc, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000000, CR3: 0x00000004577bf05e, CR4: 0x00000000003626e0
RAX: 0xffffff805560c8f0, RBX: 0xffffff8020a244c8, RCX: 0x0000000000000000, RDX: 0xffffff805560c8f0
RSP: 0xffffff9222ddba00, RBP: 0xffffff9222ddba80, RSI: 0x00000003301bd8ac, RDI: 0xffffff805560c8f0
R8:  0x000000000000005a, R9:  0x0000000000000000, R10: 0xffffff7fa0fa1e28, R11: 0xffffff7fa3211c20
R12: 0x0000000000000000, R13: 0xffffff8052498a80, R14: 0x0000000000000000, R15: 0x000000000000005a
RFL: 0x0000000000010246, RIP: 0xffffff7fa3211ccc, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x8, PL: 0, VF: 0

Backtrace (CPU 8), Frame : Return Address
0xffffff9222ddb460 : 0xffffff801ff3bb2b mach_kernel : _handle_debugger_trap + 0x47b
0xffffff9222ddb4b0 : 0xffffff80200734d5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff9222ddb4f0 : 0xffffff8020064f4e mach_kernel : _kernel_trap + 0x4ee
0xffffff9222ddb540 : 0xffffff801fee2a40 mach_kernel : _return_from_trap + 0xe0
0xffffff9222ddb560 : 0xffffff801ff3b217 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff9222ddb660 : 0xffffff801ff3b5fb mach_kernel : _panic_trap_to_debugger + 0x21b
0xffffff9222ddb6b0 : 0xffffff80206d2aa9 mach_kernel : _panic + 0x61
0xffffff9222ddb720 : 0xffffff80200652fa mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff9222ddb8a0 : 0xffffff8020064ff8 mach_kernel : _kernel_trap + 0x598
0xffffff9222ddb8f0 : 0xffffff801fee2a40 mach_kernel : _return_from_trap + 0xe0
0xffffff9222ddb910 : 0xffffff7fa3211ccc com.hieplpvip.AsusSMC : __ZN13AsusHIDDriver21handleInterruptReportEyP18IOMemoryDescriptor15IOHIDReportTypej + 0xac
0xffffff9222ddba80 : 0xffffff7fa0f54944 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice20handleReportWithTimeEyP18IOMemoryDescriptor15IOHIDReportTypej + 0x2b8
0xffffff9222ddbb10 : 0xffffff7fa0f554a0 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice18_HandleReport_ImplEyP18IOMemoryDescriptorj15IOHIDReportTypej + 0x64
0xffffff9222ddbb50 : 0xffffff7fa0f4437b com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice20_HandleReport_InvokeE5IORPCP15OSMetaClassBasePFiS2_yP18IOMemoryDescriptorj15IOHIDReportTypejE + 0x55
0xffffff9222ddbb80 : 0xffffff7fa0f44087 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9_DispatchEPS_5IORPC + 0x119
0xffffff9222ddbc40 : 0xffffff7fa0f43f68 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice8DispatchE5IORPC + 0x38
0xffffff9222ddbc90 : 0xffffff80206872c9 mach_kernel : __ZN12IOUserServer14kernelDispatchEP8OSObject5IORPC + 0x149
0xffffff9222ddbce0 : 0xffffff8020687691 mach_kernel : __ZN12IOUserServer6serverEP8ipc_kmsgPS1_ + 0x2e1
0xffffff9222ddbd80 : 0xffffff802068738c mach_kernel : _uext_server + 0x6c
0xffffff9222ddbdb0 : 0xffffff801ff41969 mach_kernel : _ipc_kobject_server + 0x1c9
0xffffff9222ddbe10 : 0xffffff801ff18635 mach_kernel : _ipc_kmsg_send + 0x135
0xffffff9222ddbe70 : 0xffffff801ff2f0e5 mach_kernel : _mach_msg_overwrite_trap + 0x2e5
0xffffff9222ddbf00 : 0xffffff802004b575 mach_kernel : _mach_call_munger64 + 0x205
0xffffff9222ddbfa0 : 0xffffff801fee3226 mach_kernel : _hndl_mach_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[F14EC4BD-41DF-30E4-9C76-03E9A4F17C24]@0xffffff7fa0f43000->0xffffff7fa0fe1fff
            dependency: com.apple.iokit.IOReportFamily(47)[988360A2-2E10-3014-A119-BE81BC045A10]@0xffffff7fa0f3c000
         com.hieplpvip.AsusSMC(1.2.2)[1EE34901-4265-31DB-BA31-931F5F28B8D8]@0xffffff7fa320a000->0xffffff7fa3224fff
            dependency: as.vit9696.Lilu(1.4.2)[3149C934-207A-31B5-A9FD-0E65D0B0AE05]@0xffffff7fa305c000
            dependency: as.vit9696.VirtualSMC(1.1.1)[6B148251-8A0B-3CB8-9D0C-31379DB7E6C9]@0xffffff7fa3085000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3D78401B-5D2D-33BC-9E41-DD2164EA874D]@0xffffff7fa09ac000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[F14EC4BD-41DF-30E4-9C76-03E9A4F17C24]@0xffffff7fa0f43000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[B7CF50DA-73BD-3233-B207-9036BCAD8E1A]@0xffffff7fa09f9000

BSD process name corresponding to current thread: AppleUserHIDDriv
Boot args: keepsyms=1 debug=0x100

Mac OS version:
19D76

Kernel version:
Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64
Kernel UUID: A8DDE75C-CD97-3C37-B35D-1070CC50D2CE
Kernel slide:     0x000000001fc00000
Kernel text base: 0xffffff801fe00000
__HIB  text base: 0xffffff801fd00000
System model name: MacBookPro15,1 (Mac-937A206F2EE63C01)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 13693050796
```

hieplpvip, 2020-03-22:
Hi everyone,

I don't have much time these days (need to prepare for my national contest), so I don't have much progress on AsusSMC. Though I think I can still help others to continue developing this kext.

After reading @williambj1 and @valinor61 and doing a quick search, I found something maybe useful.
Apple has published source code for IOHIDFamily on Catalina here: https://opensource.apple.com/source/IOHIDFamily/IOHIDFamily-1446.11.12/

The first thing I check was folder `IOHIDFamily`. Inside that you'll find there interesting files: `IOHIDEventDriver.cpp`, `IOHIDEventService.cpp` and `AppleUserHIDEventService.cpp`.

`AsusHID` is based on `IOHIDEventDriver`, which itself inherits `IOHIDEventService`. So I thought maybe `AppleUserHIDEventDriver` also inherits `AppleUserHIDEventService`. I went back and found two folders: `AppleUserHIDDrivers` and `AppleUserTestHIDDrivers`. Thanks god, `AppleUserHIDEventDriver.cpp` is indeed inside folder `AppleUserHIDDrivers`.

I had a quick look at `AppleUserHIDEventDriver.cpp` and it seems to be similar to `IOHIDEventDriver.cpp`. I don't know much about System Extensions and DriverKit, I'm still not sure what should we do with `AppleUserHIDEventDriver`.

hieplpvip, 2020-03-22:
@williambj1 
`IOProbeScore` should be integer. Anyway can you test this?
[AsusSMC-1.2.2-DEBUG.zip](https://github.com/hieplpvip/AsusSMC/files/4366280/AsusSMC-1.2.2-DEBUG.zip)

I changed IOProbeScore to 5000 (and integer, of course) and removed most of `handleInterruptReport` (since it causes kernel panic).

```patch
diff --git a/AsusHID/AsusHIDDriver.cpp b/AsusHID/AsusHIDDriver.cpp
index 8a5f9da..c405801 100644
--- a/AsusHID/AsusHIDDriver.cpp
+++ b/AsusHID/AsusHIDDriver.cpp
@@ -122,32 +122,6 @@ void AsusHIDDriver::parseCustomKeyboardElements(OSArray *elementArray) {
 
 void AsusHIDDriver::handleInterruptReport(AbsoluteTime timeStamp, IOMemoryDescriptor *report, IOHIDReportType reportType, UInt32 reportID) {
     DBGLOG("hid", "handleInterruptReport reportLength=%d reportType=%d reportID=%d", report->getLength(), reportType, reportID);
-    UInt32 index, count;
-    for (index = 0, count = customKeyboardElements->getCount(); index < count; index++) {
-        IOHIDElement *element;
-        AbsoluteTime  elementTimeStamp;
-        UInt32        usagePage, usage, value, preValue;
-
-        element = OSDynamicCast(IOHIDElement, customKeyboardElements->getObject(index));
-        if (!element || element->getReportID() != reportID)
-            continue;
-
-        elementTimeStamp = element->getTimeStamp();
-        if (CMP_ABSOLUTETIME(&timeStamp, &elementTimeStamp) != 0)
-            continue;
-
-        preValue = element->getValue(kIOHIDValueOptionsFlagPrevious) != 0;
-        value    = element->getValue() != 0;
-
-        if (value == preValue)
-            continue;
-
-        usagePage = element->getUsagePage();
-        usage     = element->getUsage();
-
-        dispatchKeyboardEvent(timeStamp, usagePage, usage, value);
-        return;
-    }
     super::handleInterruptReport(timeStamp, report, reportType, reportID);
 }
 
diff --git a/Info.plist b/Info.plist
index 8356b54..27365c2 100644
--- a/Info.plist
+++ b/Info.plist
@@ -97,6 +97,8 @@
                        <string>AsusHIDDriver</string>
                        <key>IOProviderClass</key>
                        <string>IOHIDInterface</string>
+                       <key>IOProbeScore</key>
+                       <integer>5000</integer>
                        <key>ProductID</key>
                        <integer>6246</integer>
                        <key>VendorID</key>
```

valinor61, 2020-03-22:
@hieplpvip I ve tried this in 2/3 of boots kext is not loading (Not showing any sign in IOReg) other 1/3 I am getting a panic.

hieplpvip, 2020-03-23:
@valinor61 Please send me KP log

valinor61, 2020-03-23:
@hieplpvip it seems it wasn't a panic system is just hanging because of this error

2020-03-23 08:20:33.289762+0300  localhost kernel[0]: kxld[com.hieplpvip.AsusSMC]: The following symbols are unresolved for this kext:
2020-03-23 08:20:33.298480+0300  localhost kernel[0]: kxld[com.hieplpvip.AsusSMC]: 	__ZN13VirtualSMCAPI6addKeyEjR7evectorI18VirtualSMCKeyValueXadL_ZNS1_7deleterES1_EEEP15VirtualSMCValue
2020-03-23 08:20:33.306240+0300  localhost kernel[0]: Can't load kext com.hieplpvip.AsusSMC - link failed.
2020-03-23 08:20:33.313113+0300  localhost kernel[0]: Failed to load executable for kext com.hieplpvip.AsusSMC.
2020-03-23 08:20:33.319997+0300  localhost kernel[0]: Kext com.hieplpvip.AsusSMC failed to load (0xdc008016).
2020-03-23 08:20:33.327045+0300  localhost kernel[0]: Failed to load kext com.hieplpvip.AsusSMC (error 0xdc008016).
2020-03-23 08:20:33.327734+0300  localhost kernel[0]: Couldn't alloc class "AsusSMC"

hieplpvip, 2020-03-23:
Are you using latest VirtualSMC?

valinor61, 2020-03-23:
yes I am using the latest release version.

williambj1, 2020-03-23:
@hieplpvip 

The kext you provided above didn't load on my config, I built myself one based on the git patch you provided. The result is the same, the keyboard is not responding, Apple's new driverkit driver still loads.

hieplpvip, 2020-03-23:
@valinor61 @williambj1
I'm sorry, it's my fault. I built AsusSMC with old version of VirtualSMC.
I don't know that to do now. I'm not familiar with DriverKit, and its documentation is very limited.

williambj1, 2020-03-23:
@hieplpvip 

I successfully got rid of Apple's driverkit drivers using three boot-args found in AppleUserHIDDriver's source code.

However, the keyboard is still not responding and IOReg appears exactly the same in Catalina right now. Wierd.

boot-args:
```text
AppleUserUSBHostHIDDevice-debug=0x1 AppleUserHIDEventDriver-debug=1 AppleUserUSBHostHIDDevice-blacklist
```

hieplpvip, 2020-03-23:
@williambj1 Keep those boot-args but remove AsusSMC and tell me if keyboard works.

williambj1, 2020-03-23:
@hieplpvip 

It works

hieplpvip, 2020-03-23:
I'm having a very weird problem. As I said above, when `AppleUserHIDEventDriver` is disabled, `IOHIDEventDriver` replaces it and works normally. But when I subclass `IOHIDEventDriver` in `AsusHIDDriver` it doesn't work at all.

All of the unnecessary code are removed.
```
#include "AsusHIDDriver.hpp"
#define super IOHIDEventDriver
OSDefineMetaClassAndStructors(AsusHIDDriver, IOHIDEventDriver);
```
```
#ifndef AsusHIDDriver_hpp
#define AsusHIDDriver_hpp

#include <IOKit/hidevent/IOHIDEventDriver.h>
class AsusHIDDriver : public IOHIDEventDriver {
    OSDeclareDefaultStructors(AsusHIDDriver)
};
#endif /* AsusHIDDriver_hpp */
```

valinor61, 2020-07-21:
HID Keyboard doesn't work on macOS Big Sur as well same issue

black-dragon74, 2020-07-21:
https://github.com/black-dragon74/ROG-HID

black-dragon74, 2020-07-25:
I am closing this issue as the problem is now fixed. A DriverKit based solution is now available. Users with USB HID based keyboard should use https://github.com/black-dragon74/ROG-HID

closed 2020-07-25 by black-dragon74