hieplpvip/AsusSMC #53: IOUSBHostFamily version issue on Mojave

serdeliuk, 2019-12-06:
On my Asus ROG strix G531GT I installed Mojave 10.14.6 and patched the DSDT for F7/F8 LCD backlight keys and arrow keys, the keyboard backlight and all keys works as expected with all kexts in clover/EFI

However, i cloned my old macbook's hdd which has mojave 10.14.4 on a secondary partition and when i boot on 10.14.4 AsusSMC didn't not worked, moved extensions to /L/E and nothing happen, after a kext clear cache i saw below error.

`AsusSMC.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
AsusSMC.kext is missing dependencies (including anyway; dependencies may be available from elsewhere)
`
Then i open the AsusSMC extension's config.plist and modified
`
               <key>com.apple.iokit.IOUSBHostFamily</key>
                <string>1.0.0</string>
`
to
`               <key>com.apple.iokit.IOUSBHostFamily</key>
                <string>1.2.0</string>
`
which is the right version and all keys started to work.

Now i wonder why on 10.14.6 works but not on 10.14.4 whyle the IOKIT is the same version on both macOSes

hieplpvip, 2019-12-22:
Hi, I have changed the required version from `1.0.0` to `1.0.1` since some other kexts use that version too. Can you test this?

[AsusSMC-1.2.0-DEBUG.zip](https://github.com/hieplpvip/AsusSMC/files/3992129/AsusSMC-1.2.0-DEBUG.zip)

serdeliuk, 2019-12-22:
Hi, thanks, unfortunately I have upgraded that macOS to 10.14.6 on which the RELEASE version works, the attached DEBUG version too, i have tested all 3 .configs and all works as expected.

There was something wrong with **macOS 10.14.4** i will create a new partition on my hdd and put the old 10.14.4 to test again and report the result here.

closed 2020-09-15 by hieplpvip