hieplpvip/AsusSMC #56: Help macOS 10.15 KabyLake Asus Rog GL703VD

theEvilbyteS, 2020-01-01:
Hello, I need help on my notebook i5-7300HQ, I'll try to apply KabyLake patch to DSDT but PCI0.LPCB.EC0.WRAM and PCI0.LPCB.EC0.ST9E doesn't exit. In the search of similar variables i found a method called GLKB that store KBLV but I don't know what I need to replace in the patch.
This is my Keyboard layout:
F1: Mute
F2/F3/F4: Previous/Play-Pause/Next media
F5: Fan Booster
F6: Turn off screen
F7/F8: Decrease/Increase screen backlight
F9: Switch monitor
F10: Disable trackpad
F11: Sleep
F12:Airplane mode

Please help me. Thanks.

[DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/4013712/DSDT.aml.zip)

cowpod, 2020-01-27:
that asus rog model uses an hid keyboard, not ps2/wmi/whatever, so standard or similar patches wont work; in my experience the ACPIDebug EC queries don't work. 
The Asus /HID component in this should work but doesn't- it completely breaks my keyboard.

festano92, 2020-10-10:
Did you solve? I have the same model. Keyboard backlight actually worked with clover+Mojave without patching, using fn+up and fn+down

cowpod, 2020-10-10:
take a look at the change logs.
https://github.com/hieplpvip/AsusSMC/releases/tag/1.4.1

festano92, 2020-10-10:
I tried to use ROG-hid and all blackdragon's stuffs, but it doesn't work, the backlight of my keyboard are stuck at the same color and turned on. Any advice?

cowpod, 2020-10-10:
it worked fine for me when i tried it. i dont have this device any more. make sure you follow the instructions for rog-hid - there are some extra steps involved.

festano92, 2020-10-10:
Thanks a lot!

festano92, 2020-10-11:
Lol, it’s working, thanks! But what a pain.