hieplpvip/AsusSMC #58: Work and compatible with virtualSMC 1.1.0

SkreamDark, 2020-01-13:
Today i updated VirtualSMC to 1.1.0 version and after that AsusSMC stoped working, keyboard and backlight working, but fn keys don't working, only if you install again Virtual SMC 1.0.9. Also its weird that in hackintool AsusSMC don't show on tab - installed .

rbesora, 2020-01-13:
Hi,

It just happened to me, see last closed issue. You should recompile AsusSMC using last version of Lilu and VirtualSMC and it will work. Otherwise you should wait to a new release.

Here you have the binaries I built, just in case you cannot wait.
[AsusSMC.zip](https://github.com/hieplpvip/AsusSMC/files/4054914/AsusSMC.zip)

Regards

SkreamDark, 2020-01-13:
ok, thanks), but then I'll wait, there is no need to do it right now.

feartech, 2020-01-13:
> Hi,
> 
> It just happened to me, see last closed issue. You should recompile AsusSMC using last version of Lilu and VirtualSMC and it will work. Otherwise you should wait for a new release.
> 
> Here you have the binaries I built, just in case you cannot wait.
> [AsusSMC-1.2.0-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/4054982/AsusSMC-1.2.0-RELEASE.zip)
> [AsusSMC-1.2.0-DEBUG.zip](https://github.com/hieplpvip/AsusSMC/files/4054983/AsusSMC-1.2.0-DEBUG.zip)
> 
> Regards

hey! thanks for that! worked like a charm, brightness keys now working! many thanks! :)

LeeBinder, 2020-01-20:
thanks @rbesora , very help- and useful!

I'm wondering, is there anything you might be able to fix in the code for these issues?

1. [Keyboard backlight remains off after sleep · Issue #46 · hieplpvip/AsusSMC](https://github.com/hieplpvip/AsusSMC/issues/46)

2. AsusSMC v.1.2.0 does not work on my Asus Vivobook in Mojave, only in Catalina, so I've been using v.1.1.1, which however is not compatible with the latest VirtualSMC v.1.1.0/ Lilu v.1.4.1. 

As to not get too far behind without knowing for how much longer hieplpvip might remain busy, would it be a lot of effort for you to compile [AsusSMC v.1.1.1](https://github.com/hieplpvip/AsusSMC/releases/tag/1.1.1) with the fix you already applied to v.1.2.0 above?

rbesora, 2020-01-20:
> thanks @rbesora , very help- and useful!
> 
> I'm wondering, is there anything you might be able to fix in the code for these issues?
> 
> 1. [Keyboard backlight remains off after sleep · Issue #46 · hieplpvip/AsusSMC](https://github.com/hieplpvip/AsusSMC/issues/46)
> 2. AsusSMC v.1.2.0 does not work on any of the Asus Vivobook S15 series notebooks in Mojave, only in Catalina, so I and others have been sticking to v.1.1.1, which however is not compatible with the latest VirtualSMC v.1.1.0/ Lilu v.1.4.1.
> 
> As to not get too far behind without knowing for how much longer hieplpvip might remain busy, would it be a lot of effort for you to compile [AsusSMC v.1.1.1](https://github.com/hieplpvip/AsusSMC/releases/tag/1.1.1) with the fix you already applied to v.1.2.0 above?

Hi,

Sorry, I cannot help you with #1. I have just compiled version AsusSMC v1.1.1 with  VirtualSMC v.1.1.0 and Lilu v1.4.1. I hope that works for you.

Best Regards

[1.1.1.DEBUG.zip](https://github.com/hieplpvip/AsusSMC/files/4088145/1.1.1.DEBUG.zip)
[1.1.1.RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/4088146/1.1.1.RELEASE.zip)

LeeBinder, 2020-01-21:
great, thank you @rbesora , but unfortunately no functionality in Mojave with your AsusSMC v1.1.1 with VirtualSMC/ SMCProcessor v.1.1.0 and Lilu v1.4.1 (above all no keyboard backlight).

As soon as I reapply stock hieplvip AsusSMC v1.1.1 + VirtualSMC/ SMCProcessor v.1.0.9, all is back to normal.

(yes, I always repair L/E perms + rebuild caches before reboot)

Have you compiled in Mojave, or if in Catalina, then with the 10.14 sdk present in the project?

makito89, 2020-01-26:
Hi,
I have the same issue reported here but when I attempt to use the AsusSMC @rbesora provided (1.20 and 1.1.1) I get kernel panic.
The same thing happens also with the kext I compile from current source code.
I have an Asus UX330UA with Catalina 10.15.2.
Is there a way I can debug this kernel panic ?

LeeBinder, 2020-02-01:
for me the latest 1.2.1 solves this issue - verified working in 10.13.6, 10.14.6 and 10.15.3

SkreamDark, 2020-02-01:
confirmed, working with lilu 1.4.1 and VirtualSMC 1.1.0

closed 2020-02-06 by hieplpvip