hieplpvip/AsusSMC #74: How to enable "backlight keyboard control options" in keyboard settings? (AsusSMC)

morfina, 2020-05-13:
Hello guys!
I have a Asus Zenbook with the following characteristics:
Model: ASUS Zenbook UX310UQ
CPU: Intel Core i7-7500U CPU @ 2.70GHz
IGPU: Intel HD Graphics 620
RAM: 8GBx2 => 16GB RAM Bus 2133
Wifi: Replaced with DW1560.
Clover: Clover 5117
OS: 10.15.4 (19E287)
UEFI/LEGACY: UEFI
Storage: SSD SanDisk Ultra II 240GB

Everything works fine! :)

Enable keyboard backlighting with AsusSMC (https://github.com/hieplpvip/AsusSM...Rnd0_hRK-2Eu-3FweoLJM7-sLbFS7P4I0434ZO3SInD6k). Keyboard backlight works ok, but in the keyboard settings options (system preferences) I don't see the backlight keyboard control options :(

how to show "adjust keyboard brightness in low light" in keyboard settings"?
And
how to show "Turn keyboard backlight off after xx secs of inactivity"?

these options work?
Thanks!!

Ubsefor, 2020-05-13:
Because Apple has rewritten AppleLMUController.kext which is renamed to AppleSMCLMU.kext on Catalina, AsusSMC can't provide additional backlight settings for keyboard anymore, see [this](https://github.com/hieplpvip/AsusSMC/issues/20#issuecomment-541525135) issue thread for details

morfina, 2020-05-13:
ok thanks for the info

morfina, 2020-05-13:
Is there any solution available? I saw that there is a kexts called AsusSmcCatalina.kext, does it work?

Ubsefor, 2020-05-13:
> Is there any solution available? I saw that there is a kexts called AsusSmcCatalina.kext, does it work?

Currently, **no solution available**, AsusSMCCatalina is just a debug kext used by people there for development. If it solved the problem, it would be included in some release.

morfina, 2020-05-13:
perfect! thanks for the info :)

closed 2020-08-17 by hieplpvip

rafaelmaeuer, 2020-12-27:
Is there any progress on solving this problem?

Ubsefor, 2020-12-27:
> Is there any progress on solving this problem?

It was solved long time ago with newer AsusSMC

rafaelmaeuer, 2020-12-28:
> > Is there any progress on solving this problem?
> 
> It was solved long time ago with newer AsusSMC

hmm for me its not working, I created a new issue to discuss

shiecldk, 2021-08-14:
I have same problem in here. Still cannot enable this option.