hieplpvip/AsusSMC #78: Not loading with OpenCore bootloader

Ubsefor, 2020-07-11:
Well, the Issue name speaks for itself: the kext can’t load when using OpenCore 0.6.0

valinor61, 2020-07-21:
@Ubsefor you need to recomplie this kext with latest lilu and virtualsmc kext from their sources. Only that way it will work

Ubsefor, 2020-07-22:
> @Ubsefor you need to recomplie this kext with latest lilu and virtualsmc kext from their sources. Only that way it will work

I am using correct versions of virtualsmc, wg and lilu for this kext, at least I am sure I do, will check it a bit later, as I got rid of my experimental bootable usb drive with OpenCore and decided to stick with clover.

hieplpvip, 2020-08-18:
@Ubsefor Please attach your EFI folder

Ubsefor, 2020-08-18:
> @Ubsefor Please attach your EFI folder

As a matter of fact, back then I recompiled all the kexts manually, using the latest commit of lilu, and it did work, but I got several problems with power management of my iGPU, so I reverted back to stable versions

Forgot to close this issue, sorry :(

closed 2020-08-18 by Ubsefor