hieplpvip/AsusSMC #79: Battery threshold

tunglamvghy, 2020-07-12:
I hope you can port the feature to limit the percentage of charging at 80% like My Asus app in Windows.

Vmonheros, 2020-08-07:
No need to do that, it's default feature in last version of OSX...

hieplpvip, 2020-08-24:
@tunglamvghy When the threshold is set to 80% and the battery is charged to 80%, does Windows report battery level as 100% or 80%?

tunglamvghy, 2020-08-24:
> @tunglamvghy When the threshold is set to 80% and the battery is charged to 80%, does Windows report battery level as 100% or 80%?

Trong Windows vẫn hiển thị là 80% và ngừng sạc. 
Trong MyAsus có 3 option là sạc đến 100%, 80% hoặc 60% là ngưng sạc.

hieplpvip, 2020-08-24:
Em biết anh là người Việt rồi mà do trên GitHub chủ yếu dùng tiếng Anh nên em vẫn ưu tiên tiếng Anh hơn để người nước ngoài đọc được :v Mà thôi anh đã dùng tiếng Việt nên em cũng thế luôn :v

Tạm thời em set cố định là 80% thì ngừng sạc. Anh có thể bật/tắt nó trong Battery Health.
Version này em cũng thêm support fan speed. Anh check giúp xem nó có hoạt động ko (iStats).

Do em dùng API mới của VirtualSMC (chưa release) nên anh phải update luôn VirtualSMC.

[Lilu-1.4.7-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/5116070/Lilu-1.4.7-RELEASE.zip)
[VirtualSMC-1.1.6-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/5116071/VirtualSMC-1.1.6-RELEASE.zip)
[AsusSMC-1.4.0-DEBUG.zip](https://github.com/hieplpvip/AsusSMC/files/5116122/AsusSMC-1.4.0-DEBUG.zip)
[AsusSMC-1.4.0-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/5116124/AsusSMC-1.4.0-RELEASE.zip)

tunglamvghy, 2020-08-24:
Test ok nhé bác. Chạy ổn định 😁. Mặc dù pin còn hơn 80% cắm sạc là ngừng không sạc nữa. Đèn báo từ đỏ chuyển thành trắng luôn.
[image]

[image]

tunglamvghy, 2020-08-24:
After changing from Debug to Release version, the kext doesn't work (still charging until full and not show fan speed). And I changed back to debug version, the battery threshold worked but fan speed doesn't work anymore.
[image]

Vmonheros, 2020-09-03:
Hi.

I use your new kext since you release it and no issue on my side (UX430 UAR). Battery level stop at 83%.

closed 2020-09-15 by hieplpvip