hieplpvip/AsusSMC #82: AsusSMC 1.3.0 not working with latest available VSMC, WEG and Lilu

Ubsefor, 2020-08-18:
Latest AsusSMC doesn’t work with latest commits or available for download release versions of VirtualSMC, WhateverGreen and Lilu.
This may be due to daemon not starting with launchd on my system, in which case I still need help.

hieplpvip, 2020-08-18:
Send me your EFI folder

Ubsefor, 2020-08-18:
> Send me your EFI folder

Here you go: (cleared serials)

[EFI.zip](https://github.com/hieplpvip/AsusSMC/files/5089907/EFI.zip)

hieplpvip, 2020-08-18:
Can you run [gen-debug](https://github.com/black-dragon74/OSX-Debug)? I need kernel log to see why AsusSMC is not loaded.

Ubsefor, 2020-08-18:
> Can you run [gen-debug](https://github.com/black-dragon74/OSX-Debug)? I need kernel log to see why AsusSMC is not loaded.

Here are two gen_debug outputs: one when using AsusSMC 1.3.0 and the other one with version 1.2.2

[debug_21441_ASMC130.zip](https://github.com/hieplpvip/AsusSMC/files/5090700/debug_21441_ASMC130.zip)
[debug_21517_ASMC122.zip](https://github.com/hieplpvip/AsusSMC/files/5090702/debug_21517_ASMC122.zip)

Ubsefor, 2020-08-18:
Ok, so I did a little digging in kernel logs myself, and discovered that in header file the setPowerState was declared as purely virtual method, but not implemented. So I removed declaration from AsusSMC.hpp, recompiled it and tested. The kext now loads and works correctly, I will create pull request, if you want to use it as temporarily solution (unless you wanted to implement setPowerState method in some way)

hieplpvip, 2020-08-18:
Please do it. That's function is part of the workaround for Catalina. We don't need it anymore.

closed 2020-08-18 by hieplpvip