hieplpvip/AsusSMC #85: Icelake, need help with keyboard backlight and fn keys

Paolo7297, 2020-09-04:
Hello, I have Ice Lake and i don't know how to enable the keyboard backlight.
I also have different fn keys:
f1: Mute
f2: vol -
f3: vol +
f4: screen backlight -
f5: screen backlight +
f6: disable touchpad
f7: cycle backight (4 level, after the highest -> off, then again)
f8: switch monitor
f9: ins
f10: fine
f11: pag up
f12: pag down

I also have a sort of "lock" using fn+esc (it enables a led on fn key), but it would be enough to have a swap between fn keys and f1-12 (to have the default behaviour of fn keys and with fn pressed of f1-12) if possible.
Thanks a lot! 

[01-DSDT.aml.zip](https://github.com/hieplpvip/AsusSMC/files/5177279/01-DSDT.aml.zip)

Suzamax, 2020-12-23:
Same keyboard layout here. Applying BrightnessKeys from Acidanthera too but can't modify kbd's brightness. 

About FN->Fx, it's on SysPref > Keyboard. I think if the first thing is solved the second one will be too.

btw what Zenbook do you have?

Paolo7297, 2020-12-23:
Using the patches from this repo i managed to make them working, i used the fn+f6 to lower the brightness and fn+f7 to increase the brightness.
I tried to modify the code to get the four step backlight, but it didn't worked as i wanted, so i rolled back.

to enable the lock, I don't know where to start...

unknownr0ad, 2022-04-17:
Hey man I have the same laptop, would you mind to share your configuration (EFI) folder with me so I can make my keyboard backlight works as it supposed to, I dont know anything about patching, thanks in advance

Pradanax, 2024-02-05:
@unknownr0ad are u solved thats problem?