hieplpvip/AsusSMC #86: Keyboard Backlight works with patched DSDT but not with Hotpatching

mklengel, 2020-09-13:
My Asus G750JM (Haswell, no ALC) works nearly perfect: OpenCore 6.1, Catalina.
The Keyboard Backlight works only with a patched DSDT but not with Hotpatching (no customized DSDT, only SSDT stuff).
The keys for this are Fn-F3 (decrease keyboard backlight) and Fn-F4 (increase keyboard backlight). Typing the keys lets pop up a small window so the right methods are called:

[image]

The debug messages during Fn-F3 and Fn-F4 are:

```
$ log stream --predicate "sender == \"Lilu\"" --style compact | grep atk
2020-09-13 20:08:17.188 Df kernel[0:824] (Lilu) AsusSMC       atk: @ (DBG) Dispatched key 9(0x9), loop 1 time(s)
2020-09-13 20:08:17.188 Df kernel[0:824] (Lilu) AsusSMC       atk: @ (DBG) Received key 197(0xc5)
2020-09-13 20:08:17.896 Df kernel[0:824] (Lilu) AsusSMC       atk: @ (DBG) Dispatched key 8(0x8), loop 1 time(s)
2020-09-13 20:08:17.896 Df kernel[0:824] (Lilu) AsusSMC       atk: @ (DBG) Received key 196(0xc4)
```

Clearly the called methods

```ATKD.IANE (0xC5)```

and

```ATKD.IANE (0xC4)```

doesn't work correctly in my current Non-DSDT case. May the problem lies in some inaccessible variables... may be this one (see SSDT-ATKD)?

```
 \_SB.PCI0.LPCB.EC0.WRAM (0x04B1, Arg0)
```

I created two files, SSDT-ASL0 and SSDT-ATKD:

```
// SSDT-ASL0
DefinitionBlock ("", "SSDT", 2, "ACDT", "ALS0", 0x00000000)
{
    External (_SB.ATKD, DeviceObj)
    
    Device (_SB.ALS0)
    {
        Name (_HID, "ACPI0008" /* Ambient Light Sensor Device */)  // _HID: Hardware ID
        Name (_CID, "smc-als")  // _CID: Compatible ID
        Name (_ALI, 0x012C)  // _ALI: Ambient Light Illuminance
        Name (_ALR, Package ()  // _ALR: Ambient Light Response
        {
            Package ()
            {
                0x64, 
                0x012C
            }
        })
        Method (_STA, 0, NotSerialized)  // _STA: Status
        {
            If (_OSI ("Darwin"))
            {
                Return (0x0F)
            }
            Else
            {
                Return (Zero)
            }
        }
    }
    
    Scope (_SB.ATKD)
    {
        Method (ALSS, 0, NotSerialized)
        {
            Return (\_SB.ALS0._ALI) /* External reference */
        }
[... 254 lines cut ...]
        Method (_Q72, 0, NotSerialized)  // _Qxx: EC Query, xx=0x00-0xFF
        {
            \RMDT.P1 ("ACPIDebug: EC _Q72 enter")
            If (\_SB.ATKP)
            {
                \_SB.ATKD.IANE (0x82)
            }

            \RMDT.P1 ("ACPIDebug: EC _Q72 exit")
        }

        // Fn V
        Method (_Q74, 0, NotSerialized)  // _Qxx: EC Query, xx=0x00-0xFF
        {
            \RMDT.P1 ("ACPIDebug: EC _Q74 enter")
            If (\_SB.ATKP)
            {
                \_SB.ATKD.IANE (0x8A)
            }

            \RMDT.P1 ("ACPIDebug: EC _Q74 exit")
        }
    }
}
```

hieplpvip, 2020-09-15:
Hi, since you have marked this issue as solved, I will close it.

Just a little remark: `ALSS` and `ALSC` is not needed on v1.4.0. You can safely remove them.

closed 2020-09-15 by hieplpvip

mklengel, 2020-09-15:
Thank you for the information and - of course - your invaluable work.

Regards
Michael

„Don’t talk about nothing you know nothing about.“ John Young