hieplpvip/AsusSMC #87: AsusSMC could not be loaded

Ubsefor, 2020-09-15:


Ubsefor, 2020-09-15:
AsusSMC 1.4.0 could not be loaded by Clover 5120, due to LiluFriend doesn’t load

Ubsefor, 2020-09-16:
The problem also persists on Clover 5122

hieplpvip, 2020-09-16:
OpenCore works fine for me

lathoskaneis, 2020-09-17:
I have also the same problem with Clover tested on 5117,5122.

hieplpvip, 2020-09-17:
Could you guys dump boot log?

lathoskaneis, 2020-09-17:
Do you need the whole log?

lathoskaneis, 2020-09-17:
[image]

lathoskaneis, 2020-09-18:
Can I help more?Do you need full log or can I try something else?
Thank you for your time Hiep.

bernardo415, 2020-09-20:
I want to test with opencore on my asus X540LA but not being able to use DSDT in OC as I can use this kext and patch what it takes to make it work?

Ubsefor, 2020-09-20:
It works fine with OC, as OC doesn’t need _LiluFriend.kext_ for AsusSMC to work properly
Also, there’s a branch in this repo with configs for OC, I will soon create a merge request to fix a config for UX430UAR, as the sleep doesn’t seem to work with provided one, but I fixed it on my machine

hieplpvip, 2020-09-22:
Please test this
[AsusSMC-1.4.1-RELEASE.zip](https://github.com/hieplpvip/AsusSMC/files/5261197/AsusSMC-1.4.1-RELEASE.zip)

lathoskaneis, 2020-09-22:
For me seems to load correctly, everything works, keyboard backlight, automatic brightness, fn+ keys.
Just to let you know that I have never used lilufriend in order to load your kext (and still I am not using it)
My laptop is an UX410UN with Intel 8250U without nvdia gfx.
Thank you for your work.

sj64, 2020-10-25:
Prob not very helpful - just thought I'd share my experiences with this excellent kext.

Hardware: Asus UX310UA 2.9GHx i7 24GB ram
MacOs : Catalina 10.15.7
Bootloader: rEFInd chain loading opencore just for MacOs (Quadboot all UEFI system)
All kexts latest versions = AsusSMC 1.4.1

Had this kext working perfectly with 10.15.4 (Clover).
Different story with current system.
None of the suggested alternatives work for me Labtick/kbrightness etc. Every one of them reports unable to connect to AppleLMUController, or no keyboard backlight detected.

Had system running ok without patched DSDT as per opencore recommendation, but no KB backlight, and trackpad had only very basic functionality (move and click only, no tap even), and was not detected in system preferences.

Decompiled and patched DSDT solely for this kext, applied brightness patch first then reboot to see what works, applied patches to f3,f4,f5,f6, reboot and screen brightness works, but still no keyboard backlight. HOWEVER mysteriously now my trackpad (ELAN1000 I2C) was detected AND fully functional - gestures etc.

But still no luck with backlight, even with daemon installed and running. Sometimes soft booting from other OS with backlight on will enable it to stay on in mac, but if it sleeps it kills the backlight - so at least it can turn it off! Prob some global sleep everything command.

As a just to see, recompiled kbrightness.c to search for AppleSMCLMU instead of AppleLMUController as per Hiep's comment earlier, but of course nothing so simple - same error.

Does seem to lie deeper at the hardware detection level.

Done a fair bit of programming in other languages, and willing to learn - and I miss my backlit keys a lot! Don't know if I can help at all, but have spare time, so please point me in a direction you think might be useful and I'll see if I can make any actual progress.