hieplpvip/AsusSMC #91: Big sur 11.1

carlosgaros, 2020-12-17:
Hi support big sur 11.1 ?

Ubsefor, 2020-12-17:
All is working correctly, check the install instructions

carlosgaros, 2020-12-17:
look logs and find only this : 
Dec 17 10:43:33 MacBook-Pro-kamil com.apple.xpc.launchd[1] (com.hieplpvip.AsusSMCDaemon): This service is defined to be constantly running and is inherently inefficient.
Dec 17 10:43:33 MacBook-Pro-kamil com.apple.xpc.launchd[1] (com.hieplpvip.AsusSMCDaemon): The ServiceIPC key is no longer respected. Please remove it.

Ubsefor, 2020-12-17:
Please install the latest version of AsusSMC Daemon using provided scripts. The kext inself should work just fine for basic brightness and volume management, providing you have the standard asus's fn-keys layout (if not, you have to patch your DSDT to map those keys accordingly)

carlosgaros, 2020-12-17:
> Please install the latest version of AsusSMC Daemon using provided scripts. The kext inself should work just fine for basic brightness and volume management, providing you have the standard asus's fn-keys layout (if not, you have to patch your DSDT to map those keys accordingly)
> Also check, that you have the latest Lilu/WEG/VirtualSMC

Check all . All fnk key work sleep, wifi, backlight lcd,sound,disable lcd , only keyboard light not work ....

carlosgaros, 2020-12-19:
Thanks .Now all work .asus n550jk not have light sensor after patch fakeals work perfekt :)

closed 2020-12-19 by carlosgaros