hieplpvip/AsusSMC #95: ROG GX531 Keyboard backlight does not work

Eaxyx, 2021-02-15:
[image]
Although the backlight option appears in the system, I can't adjust the brightness of my keyboard through this option

Eaxyx, 2021-02-15:
Please tell me how to modify DSDT to make it work

Eaxyx, 2021-02-15:
^^KBLV = Arg0 / 16，I can't find this description in my DSDT

Ubsefor, 2021-02-15:
> ^^KBLV = Arg0 / 16，I can't find this description in my DSDT

Please, stop spamming in the issues, consider writing one big comment instead of a huge amount of small ones.
Also, please check the instructions in repo itself, they are quite self-explanatory.

There is a whole section about ACPI patching for correct brightness controll. Also, make sure that daemon is running. 

This should work. Otherwise, we need logs anyway, IOReg dump at least.

Eaxyx, 2021-02-17:
I'm sorry, but I'm just a person who is ignorant about this project. Please forgive my irrationality. I don't know where to find the modification of ACPI. If you can, please give me a link about the repair of ACPI. Thank you for your help. Here's my DSDT
[
[DSDT-GX531GS.zip](https://github.com/hieplpvip/AsusSMC/files/5995891/DSDT-GX531GS.zip)
](url)

Ubsefor, 2021-02-17:
> I'm sorry, but I'm just a person who is ignorant about this project. Please forgive my irrationality. I don't know where to find the modification of ACPI. If you can, please give me a link about the [...]

Here you go: [link](https://github.com/hieplpvip/AsusSMC/wiki/Installation-Instruction)

Basically, to remap the buttons you should change the keycode to different one (see classic zenbook keyboard layout to get some ideas about what buttons need to go where) in your ACPI. 

For example, you need to change 0x5E opcode [here](https://github.com/hieplpvip/AsusSMC/blob/master/patches/f1.txt) (after applying patch to your ACPI) to different one, for example 0x7D as it is [here](https://github.com/hieplpvip/AsusSMC/blob/master/patches/f2.txt), if your fn-F1 key is not sleep mode toggling but airplane mode. 

There once was a guide about these buttons in the «to do» kind of sections of wiki, but I can’t find it for some reason. Hiep probably deleted it(

TL;DR
Use MacIASL, add `http://raw.github.com/hieplpvip/AsusSMC/master` as source, patch your ACPI, replace opcodes that define keyboard keys to different ones.