hieplpvip/AsusSMC #97: Dependencies not found for kexts AsusSMC

andreademasi, 2021-04-17:
Hi everyone, first of all thank you for this project, i'm trying to make backlight and battery health charging working but it seems that i can't manage to get AsusSMC properly loaded:
```
Dependency as.vit9696.Lilu was not found for kext com.hieplpvip.AsusSMC
Dependency as.vit9696.VirtualSMC was not found for kext com.hieplpvip.AsusSMC
```
Any ideas why AsusSMC can't find these kexts? I do have them installed
My laptop is a ZenBook UX433FN, let me know if you need more details.

Ubsefor, 2021-04-17:
Prob. incorrect load order or kexts version, please check that. Lilu should be loaded before AsusSMC. I suggest following my kext load order:

[image]

andreademasi, 2021-04-17:
Ops, you're right, totally missed that thanks.

closed 2021-04-17 by andreademasi