hieplpvip/AsusSMC #98: Keyboard Backlight not working Asus N752VX

adultano, 2021-04-22:
Hi - I struggle with my keyboard backlight. I have an Asus N752 laptop with Skylake Intel Core i7-6700HQ and Intel HD 530 1920 x 1080 IGPU. I use OpenCore and Big Sur. Everything else works, e.g. screen brightness with FN Keys and all other FN Keys. I dual boot with Windows 10, there keyboard backlight works. I also fake ALS since it's not built in. In Keyboard settings I see backlight available. I unticked "adjust brightness in low light" and the scroll bar is available in command center. Could someone support me? Let me know what further information / files are needed.
Thanks in advance.

adultano, 2021-04-23:
Found a the solution meanwhile. Works with Big Sur

closed 2021-04-23 by adultano

Ubsefor, 2021-04-23:
Please, do not just close the issue without describing how you managed to fix it. It could help others with the same problem.

adultano, 2021-04-23:
Sure - sorry. Let me try to explain then. My system was already working fine, Dual Boot Win 10 and Big Sur through OpenCore. Except keyboard brightness everything was working. I tried to dig deeper and failed a little over two days now. LabTick didn't work, Karabiner didn't work although my keyboard was recognized to be compatible (in system preferences -> keyboard, I had the brightness options available). I also tried additional kexts like AppleBacklightInjector. No success. The solution in the end was to add another SSDT patch (see below) and map the correct number for CI0.LPCB.EC0.WRAM **(0x04B1**, Arg0). This I gathered through MaciASL under SLKB. I stored this in .aml in ACPI in EFI and made sure it gets added in config.plist. I have a fully functional 16 step keyboard backlight now.

*******.aml which i added****

_DefinitionBlock ("", "SSDT", 2, "hack", "AsusSMC", 0x00000000)
{
    External (_SB_.ATKD, DeviceObj)    // (from opcode)
    External (_SB_.PCI0.LPCB.EC0_.WRAM, MethodObj)    // 2 Arguments (from opcode)

    Scope (_SB.ATKD)
    {
        Method (SKBV, 1, NotSerialized)
        {
            ^^PCI0.LPCB.EC0.WRAM (0x04B1, Arg0)
            Return (Arg0)
        }
    }
}_

Hope this helps....