hieplpvip/AsusSMC #102: Single FN Key for Keyboard Backlight (Comet Lake Laptop)

astro-lee-k, 2021-08-06:
Hi my laptop (ASUS ZenBook Pro Duo 15 UX582) has only one key for keyboard backlight control. I've check the guide for SSDT/DSDT; however, it did not work.

I attached my Debug files. 
https://drive.google.com/file/d/1fB-K3Dw2CISDaLb6qZmpKOe4xM14fe5_/view?usp=sharing

Similar issue:
https://github.com/hieplpvip/AsusSMC/issues/101

shiecldk, 2021-08-12:
@Ubsefor @hieplpvip Could you help?

Ubsefor, 2021-08-12:
> @Ubsefor @hieplpvip @AR-CADE Could you help?

Check #93 and #95

shiecldk, 2021-08-28:
@Ubsefor @hieplpvip @AR-CADE I tried but the Keyboard Backlight FN key doesn't work.

robertr1229, 2021-09-18:
Same issue with Ice Lake. Laptop. I have tried multiple mappings without success to get the f4,f5, and f7 keys to function properly. The link to custom mappings in the wiki is dead with a page that says "to do"

shiecldk, 2021-09-19:
@robertr1229 I actually got the FN key for volume, brightness, and keyboard backlight work on my Ux582 laptop. Check my repo for ASUSFN.aml and AsusSMC.aml.

https://github.com/shiecldk/ASUS-ZenBook-Pro-Duo-15-OLED-UX582-Hackintosh/tree/main/EFI/OC/ACPI

pabbiebottoman, 2023-04-29:
Hi @shiecldk ,  Can you please help me with something?