hieplpvip/AsusSMC #103: Resume from hibernation turns off battery threshold at 80 percent

whatnameisit, 2021-09-19:
On normal use, AsusSMC 1.4.0+ caps the battery level at 80 percent if `Battery Health` in `Energy Saver` is toggled on.
When the machine resumes from hibernation, the threshold is ignored and the battery keeps on charging until 100 percent.
I'm not good with logging (and about zero knowledge in coding), so if anyone could tell me how, I'll collect the logs and upload them here.

Ubsefor, 2021-09-19:
This feature shouldn't work at all on non-apple hardware. I tried to use it for a bit and noticed that my battery doesn't charge correctly with this turned on: battery dies at 30-40% immediately. 

I don't know whether you have the same issue or not, but I suggest at least checking it out.

whatnameisit, 2021-09-19:
I'm not sure we are on the same issue. By "this feature," did you mean battery threshold on hibernation or just the battery threshold?
Hibernation works with HibernationFixup.kext and battery threshold works with AsusSMC.kext.
When both kexts are used together, battery threshold is ignored when resuming from hibernation. A simple reboot makes the limit to kick in and prevent the charging above 80 percent again.
I don't remember experiencing the laptop shutting down at 30-40 percent battery level, if that's what you mean by `battery dies at 30-40% immediately.` I'm pretty sure there were times the battery level was below 25 percent and it charged fine with the battery threshold feature on.
Since you commented, do you know how to debug or collect logs on my issue?

yjmd2222, 2024-07-20:
>1.4.0
>Support battery charge threshold found on new laptops (charging stops at 80%). Can be disabled by turning off Battery Health in Energy Saver.

Can it really be turned off in Energy Saver in Catalina?
https://github.com/hieplpvip/AsusSMC/blob/c15d2f297c86b65a3cb866cc2bcece595c136417/AsusSMC/AsusSMC.cpp#L379-L390
https://github.com/hieplpvip/AsusSMC/blob/c15d2f297c86b65a3cb866cc2bcece595c136417/AsusSMC/AsusSMC.cpp#L604-L616
https://github.com/hieplpvip/AsusSMC/blob/c15d2f297c86b65a3cb866cc2bcece595c136417/AsusSMC/AsusSMC.cpp#L871-L875

I don't see any reading from Battery Health that is used to turn it off.

Anyways, I think the original issue can be resolved if the kext calls the method on resume. I don't know the details, but hibernation is similar to shutdown, which if true will reset EC to original, non-conservation mode. A quick check is to call the corresponding ACPI method `EC0.SRSC (0)` on resume with ACPIDebug.kext.

yjmd2222, 2024-08-03:
I'm blind
https://github.com/hieplpvip/AsusSMC/blob/c15d2f297c86b65a3cb866cc2bcece595c136417/AsusSMC/KeyImplementations.cpp#L55-L67

I can't get the machine to hibernate, so can't test it. Tried: HibernateMode=Auto/RTC/NVRAM, ThirdPartyDrives=True, HibernationFixup.kext=enabled, and hibernatemode=25/3.
If anyone has step-by-step directions to enable hibernation, please share, because I might have missed something.

yjmd2222, 2024-08-07:
Upon resume from hibernation, the battery does charge above the threshold. Then I set the threshold with ACPIDebug.
```
[ 1267.932184]: [ACPI Debug]  [ACPI Debug]  "Setting SRSC = 80"
```
And the battery stops charging.
[image]
The icon shows AC is plugged in and not charging.

The kext code can be updated to send battery health command on resume.