hieplpvip/AsusSMC #104: I want asussmmc.kext to charge until 59% only :D

ericgaldeano, 2021-11-29:
I want to congratulate you for your amazing work, I am a student in Italy and I am amazed of your work. I did install on my Asus X411UF Monterey hackintosh thanks to the pack : https://github.com/tunglamvghy/AsusS530UN-hackintosh  that i am pretty sure includes the asussmc.kext file because this pack limit the battery charged to 80% more or less although mine is 83%. In my windows I have set the maximum lifespan for my battery so only charges to 58% :D. So about the tge  "battery with battery threshold feature" on the asussmc u made I would like to reduce it to 58% , how it can be done? Do I need to change a simple value or is difficult than that ? Could you help me witht this file ?Thank you so much teacher!! And a happy day

Bhavinjain260, 2022-06-02:
yes it can be done

Bhavinjain260, 2022-06-02:
[image]
Replace 80 with 59 or 57 
and compile the kext

Bhavinjain260, 2022-06-02:
the files is in this location 
AsusSMC > AsusSMC.cpp