hieplpvip/AsusSMC #105: SSDT patching?

nvmnghia, 2021-12-11:
I read it somewhere on the Dortania guide that you shouldn't put `DSDT.aml` on ACPI (but it didn't say `SSDT` was the only option neither). Is there any way to just do `SSDT` patch, instead of patching the whole `DSDT`?

hamcuks, 2022-01-02:
Yes, you can patching EC Query without DSDT. Just rewrite the EC Query on SSDT, and add Rename the Original EC Query on config.plist

Here is the example: https://github.com/hamcuks/hackintosh-x442urr/blob/master/EFI/OC/ACPI/SSDT-X442UR.dsl

nvmnghia, 2022-01-03:
Hi, this is a bit overwhelming for me @hamcuks . Is there any beginner guide on ACPI language?

hamcuks, 2022-01-05:
> 

AFAIK, It's the simple way if wanna using SSDT.

Just follow the steps:
1. Rename the original EC Query from DSDT (Example: _Q0E to XQ0E). Rename using find and replace in config.plist
2. Rewrite the modified EC Query in SSDT

18iteration, 2023-05-01:
> Yes, you can patching EC Query without DSDT. Just rewrite the EC Query on SSDT, and add Rename the Original EC Query on config.plist
> 
> Here is the example: https://github.com/hamcuks/hackintosh-x442urr/blob/master/EFI/OC/ACPI/SSDT-X442UR.dsl

The site you linked to, containing the dsl file, is 404 / down. There's also no copy on the Internet Archive sadly.

hamcuks, 2023-05-01:
> The site you linked to, containing the dsl file, is 404 / down. There's also no copy on the Internet Archive sadly.

I've archived my repo, actually. The repository is no longer maintained

DotCube123, 2023-07-27:
Is there any easy way to do it? Or at least guide on hotpatching SSDT's? Kinda nerving me that I am unable to do anything when it comes to patching these kind of stuff.