hieplpvip/AsusSMC #107: Some FN keys and keyboard backlight not working on Asus Vivobook Flip 14 TP412F.

imrtoyi, 2022-06-15:
Hello!
For several days I tried to get my keyboard backlight and some fn keys to work, but I did not succeed, so I'm asking for help here. There was a post in 2019 with a similar issue, it was marked as #TODO, but apparently the issue was not resolved. Below I will write all the information that I could find out, I hope it will help in solving the problem. Thanks in advance!
1. Notebook Information
Laptop model: Asus Vivobook Flip 14 TP412F
Processor: Core i3 8145U (Whiskey Lake)
2. Keyboard layout:
    1. F1 / Mute / Don't generate EC Event (ACPIDebug) or any PS2 keycode
    2. F2 / Vol Down / Don't generate EC Event (ACPIDebug) or any PS2 keycode
    3. F3 / Vol Up / Don't generate EC Event (ACPIDebug) or any PS2 keycode
    4. F4 / Brightness Down / Generate PS2 keycode
    5. F5 / Brightness Up / Generate PS2 keycode
    6. F6 / Toggle Touchpad / Don't generate EC Event (ACPIDebug) or any PS2 keycode
    7. F7 / Keyboard backlight / Don't generate EC Event (ACPIDebug) or any PS2 keycode
    8. F8 / Change video output / Generate PS2 keycode
Also on the laptop there is a volume button on the side, they work. The shutdown button works strangely when I assign a shortcut in the system preferences, when you click on it, the computer opens the shutdown menu, if the system preferences are closed, then nothing happens. A long press will force the laptop to shut down.
￼
[image]

3. Non-standard DSDT for ASUS laptops.
In this laptop, as I understand it, the _Qxx methods refer to EC with the name H_EC, while, as usual with ASUS laptops, EC has the name EC0. That is why the patches suggested for MaciASL are not suitable. In addition, a non-standard method for changing the keyboard backlight (1 button) is used. Apparently, this method is responsible for this.
￼
[image]

4. ASUSSMC finds my ATKD device.
￼
[image]

5. For some reason, EC Events are not generated when Fn keys are pressed. Maybe I did something wrong, but with ACPIDebug running and the necessary patches, when pressing the fn keys, there were no changes in the logs. At first I thought that the problem was in ACPIDebug, but I tried to apply the same to other methods, for example, related to sleep work, the corresponding data appeared in the log, so I think that the problem is with H_EC or ATKD.

I suspect that the problem may be related to the fact that Kext does not provide hardware, like mine, so I ask you to help me, since my knowledge in writing kexts and ACPI is not so great. If I can help in any way, I'm ready, thanks! 

[Logs, EFI, ACPI.zip](https://github.com/hieplpvip/AsusSMC/files/8910907/Logs.EFI.ACPI.zip)

imrtoyi, 2022-06-16:
Keyboard backlight and fn keys also works in Ubuntu. acpi_listen:
[image]
If i turn on the backlight in windows or linux the backlight level is saved on macos boot and the backlight turns off after waking up from sleep

z0rielle, 2026-02-02:
Somebody?