hieplpvip/AsusSMC #108: Always Active FN Key (or Invert FN Key)

astro-lee-k, 2022-10-16:
Hello,

I'm using AsusSMC with UX582 on macOS Monterey. Most FN keys work well with AsusSMC; however, I feel it annoying that I always need to hold FN key + function keys change volume, brightness, etc. I wonder if it is possible to put the FN key always on, just as how it can be done in Windows and Linux, without the need to always hold FN key? Thanks!

wern-apfel, 2022-12-03:
I have a working FN key lock, can you please upload your DSDT?

1alessandro1, 2022-12-24:
Same issue here, the wiki is unfortunately empty...
I'd like to be able to lock FN either via SSDT or by making the ESC + FN combination to work and swap FN with the usual function keys to adjust brightness and volume...

wern-apfel, 2022-12-25:
Are you talking about the UX582? If so, can you please upload the DSDT?

1alessandro1, 2022-12-30:
I'm talking about the F515-JA
[DSDT.dsl.zip](https://github.com/hieplpvip/AsusSMC/files/10324966/DSDT.dsl.zip)

wern-apfel, 2023-02-01:
@1alessandro1 I think you are using an Acidanthera AWAC SSDT, if so replace it with this. atm. I can only give you my old fix, it is not restored after waking up, but can be reactivated.
[SSDT-AWAC-DISABLE.aml.zip](https://github.com/hieplpvip/AsusSMC/files/10559966/SSDT-AWAC-DISABLE.aml.zip)

shiecldk, 2023-02-01:
@wern-apfel Did you solve the FN key lock problem with just the NWAK fix in the OpenCore ACPI patch? I described the problem [here](https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/issues/21#issuecomment-1412120269).

shiecldk, 2023-02-01:
Also, that DSDT is exported from macOS with some ACPI names already patched:
https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/issues/21#issuecomment-1406677288

wern-apfel, 2023-02-08:
> @wern-apfel Did you solve the FN key lock problem with just the NWAK fix in the OpenCore ACPI patch? I described the problem [here](https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/issues/21#issuecomment-1412120269).

I'm sorry, you ask me for help, but at the same time you ask others for money, and you give someone else credits for my work.  I don't think that's fair. Please understand if I don't offer you any help then.

shiecldk, 2023-02-11:
@wern-apfel What do you mean I ask others for money? Do you mean the donation? No one is paid to support the UX582 project; ppl in the UX481 project took/forked my work directly. I did not know the SSDTs were created by you; I've been busy for the whole year. I'm maintaining the project to help others. What you accused here is mean. I can of course add you to the credit list for the UX582 project. You can just remind me that I forget to give you the credit without being so toxic. In fact, almost all the projects with SSDT are not giving credit to ppl who originally created the SSDTs; you're being ridiculous. Projects on GitHub are open source. Remember the "Many Others" in the credit list in UX582.

I've been busy with my work and I'm updating the project to help others who want to use and continue the project; I do not have more time to work on the UX582 hackintosh project, and even mostly use the laptop in Windows. Only booting to macOS to help those who want to continue the project. If you want to be so mean, you should not use @Qonfused's project because it was based on my UX582 project.

shiecldk, 2023-02-11:
On top of that, if it was about the credit, I only pushed a none-fully-working EFI to the UX582 project; I have to ensure everything works fine with the sleep and others with the new SSDTs before officially announcing that the FN-keys are fully working. It is easy to lose track to update the credit list in the README.md file after haven't updated but pushing so many changes to the project. I've been occupied with other important stuff in the whole me after the last time I pushed the changes to the project. I can of course put you on to the credit once I ensure every problem is solved; before then, it's a none-fullying working case, and should be withdrawn from the project if it breaks the sleep.

Qonfused, 2023-02-11:
Attribution is preserved in https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/pull/19 (as source aml does not preserve line history), which I have referenced and described to wern-apfel before. The referenced conversations by @shiecldk mention this by name, linking wern-apfel's original w/ my derived work (this is mostly in integration details). I also do not see any attribution given to me in this conversation to suggest these claims. However, other works that are substantially transformative would be accredited to me, which motivated the inclusion of references and attribution in source SSDTs to gvtk, the bumblebee project, dortania, etc.

wern-apfel, 2023-02-11:
> On top of that, if it was about the credit, I only pushed a none-fully-working EFI to the UX582 project for those who want to boot macOS Ventura and complain the project hasn't been updated for a while; I have to ensure everything works fine with the sleep and other stuff with the new SSDTs before officially announcing that the FN-keys are fully working. It is easy to lose track to update the credit list in the README.md file after I haven't updated the UX582 project for a while but pushed so many changes to the project. I've been occupied with other important stuff for the whole week after the last time I pushed the changes to the project. I can of course put you on to the credit once I ensure every problem is solved; before then, it's a non-fullying working case and should be withdrawn from the project if it breaks the sleep. What I used was pulled from @Qonfused's fork, which may be modified based on your work as I don't see the SSDT codes being the same. He did a good job maintaining the UX481 project (also not putting you on the credit in his README.md), not sure why you're targeting me...

This is offtopic, and a pointless discussion.
Just to clarify, I don't use anything from your EFI, because I have a different model. 
About the credits: I can't remember where, but you give Qonfused credits for fixing the Screenpad backight, which is not true. 
And I still say, please do not expect help if you expect money from others. 
No more comments from me about this topic.

Qonfused, 2023-02-11:
@wern-apfel Please refrain from making accusations and starting a flame war in a Github issue. This violates Github's acceptable use site policy, and continued behavior can elevate into harassment and will result in an account suspension.

For those subscribed to this issue, please refer to https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/issues/6. This is what is intended to be incorporated as part of asus-wmi-screenpad and AsusSMC, and what is to resolve EC methods for FN+Lock and keyboard/screenpad backlight.

shiecldk, 2023-02-11:
> This is offtopic, and a pointless discussion. Just to clarify, I don't use anything from your EFI, because I have a different model. About the credits: I can't remember where, but you give Qonfused [...]

@wern-apfel You should point out where I give credit to him then. And if you point it out, I assume you've been following and referring my UX582 project.